ArduCAM/Arduino #242: Changing Mini-5MP-Plus FPS

DallonTG, 2017-07-25:
Hi,

I have several Arducam Mini-5MP-Plus units. I want to capture images until the frame buffer is full, but I want to have some control over when each frame is captured. 

It seems to me that I just need to change the fps settings of the OV5462 in order to do this. Is this assumption correct? What registers do I need to change over I2C to change the fps? 

I am on a tight schedule so I would really appreciate some help with this. Thanks.

supprot, 2017-07-26:
The continuous capture mode until the buffer is full is useful for video clip recording. 
So it is not suggested to use it as time elapsed capture.
Tunning the OV5642 register is not easy to change the fps settings, you have to add dummy lines and cause over exposure and motion blur.
You'd better capture a image a time if you take time elapse images.

DallonTG, 2017-07-26:
Capturing a single frame at a time is not an option. The problem is that I need to capture several frames and then read them out from the buffer at a later time.

I tried to do this, but for reasons that do not make sense to me, your CPLD automatically resets the FIFO write pointer before every new capture. Because of this, each new frame capture overwrites the previous frame.

Unless there is a way to stop the CPLF from resetting the FIFO write pointer automatically, I need to change the sensor's FPS.

supprot, 2017-07-26:
I'm afraid that the default CPLD firmware can't do that way.
You might need customized design for that purpose, contact us: admin at arducam dot com if you needed.

DallonTG, 2017-07-26:
I don't understand why you would build in this restriction.

I simply do not have time to wait for you to change the firmware on the CPLD so I need to change the FPS. can you provide any assistance on this?

DallonTG, 2017-07-26:
Can you tell me what frequency you are driving XVCLK (the system input clock) with is?

supprot, 2017-07-26:
The XVCLK now is 25MHz.
We upload a debug tool to adjust the sensor register on the fly, please check the following link:
www.arducam.com/downloads/app/ArduCAM_CMD_debug.rar