ArduCAM/Arduino #243: Arducam OV5642 Mini 5MP Plus not working host app or examples

chrismunley, 2017-07-28:
Hello, I am trying to use an OV5642 Mini 5MP Plus with an arduino uno/r3. I downloaded the library, put the contents of arduino-master in my library folder, and uncommented my hardware in the memorysaver.h file. When I try to use the ArduCAM_Host_V2 to take pictures, I never get any pictures, it only appears black. I am not sure if I am even connecting to the camera through the correct com port at this time. I am also pretty sure the wiring is correct, and I have tried setting the CS pin on 7 and 10. When I try to run an example code it says a ton of variables are not declared in scope ex:

ArduCAM_Mini_5MP_OV5642_Plus_Functions:624: error: 'Exposure10_EV' was not declared in this scope

    myCAM.OV5642_set_Exposure_level(Exposure10_EV);temp = 0xff;

                                    ^

ArduCAM_Mini_5MP_OV5642_Plus_Functions:627: error: 'class ArduCAM' has no member named 'OV5642_set_Exposure_level'

    myCAM.OV5642_set_Exposure_level(Exposure13_EV);temp = 0xff;

          ^

ArduCAM_Mini_5MP_OV5642_Plus_Functions:627: error: 'Exposure13_EV' was not declared in this scope

    myCAM.OV5642_set_Exposure_level(Exposure13_EV);temp = 0xff;

I am not sure what else to try... I am just trying to get a picture from this camera onto my PC at this point so I can work from there. Let me know if you have any suggestions or advice. Thanks!

UCTRONICS, 2017-07-30:
@chrismunley 
Hi,
Sorry for our library's bug and we have update it, please give us your email adress and we will send you our new library.
Let us know if you need more help.
Regards,
ArduCAM support team.

chrismunley, 2017-07-30:
chrismunley1@gmail.com

closed 2017-08-02 by chrismunley