ArduCAM/Arduino #245: Using two cameras on Raspberry Pi 3 model B

hshbnian, 2017-08-02:
I want to capture a stereo image with 2 raspberry pi camera. Hence, I'm using Raspberry Pi 3 B (I bought from Cana Kit) and a multi camera adapter module for Raspberry Pi( I bought from Arducam).
[image]
I mounted two cameras to the "Camera A" and "Camera C" ports (on the top of multi camera board). Based on the camera selection configuration, which has been mentioned here, I only need to work with three pins of GPIO( Pin# : 7,11, and 12).

At first, I wanted to make sure both cameras were working separately and simultaneously. So, before using the multi camera adapter board, I checked each camera separately(I mounted them on the raspberry Pi). Both were working correctly.

In addition, after using the multi adapter board, I checked them out separately. They were working correctly(for example, at first I mounted camera A and ran following code to capture an image, then unmounted the first camera from "camera A" port and used the second camera on the "camera c" port and took an image, they were working correctly).

However, the problem is :

When two cameras are on the board simultaneously, the first camera, which was plugged in "camera A" port, does not capture the correct image. Meanwhile, the second camera which was plugged in "camera c" works correctly.

Specific python commands for "camera A" as follows (pin 7= 0, pin 11=0 pin 12=1):

import RPi.GPIO as gp
gp.setup(gp.BOARD)

gp.setmode(7, gp.OUT)
gp.setmode(11, gp.OUT)
gp.setmode(12, gp.OUT)

gp.output(7, False)
gp.output(11, False)
gp.output(12, True)

cmd="raspistill -o capture_first.jpg"
os.system(cmd)

The captured image contains so much noise and artifacts and after showing this, the system would crash).

[image]

for the other camera, the code is same. The only thing I changed was the pins' values(pin 7= 0, pin 11=1 pin 12=0)
I would appreciate it if someone would help me. Thank you.

UCTRONICS, 2017-08-03:
@hshbnian
Hi,
Please  attached us all of your code and  we will check it for you.
Regards,

hshbnian, 2017-08-03:
@UCTRONICS 
Hello,
Thank you for your response.
I've attached three codes (MainCode.py, FirstCamera.py, and SecondCamera.py).
All codes are so simple, including:
MainCode.py==> Goal is switching between two cameras
FirstCamera==> Goal is capturing image with the first camera which is mounted on "Camera A" port
SecondCamera==> Goal is capturing an image with the second camera which is mounted on "Camera C" port.

Between these three codes, only "SecondCamera.py" is working. In other two codes, it only shows something that is completely noise.
[Codes.zip](https://github.com/ArduCAM/Arduino/files/1198100/Codes.zip)

I think that I need to change something in "config.txt", I've attached this file also.
[config.txt](https://github.com/ArduCAM/Arduino/files/1198113/config.txt)

Thank you.

UCTRONICS, 2017-08-04:
@hshbnian
Hi,
Please refer to this code
http://www.arducam.com/downloads/modules/RPI_MULTICAM/multi_capture_A.zip
Regards.

hshbnian, 2017-08-04:
@UCTRONICS
If you look at my code, you would see that I used a code that you mentioned. Camera A does not work correctly when we have two cameras simultaneously on the board.

UCTRONICS, 2017-08-07:
@hshbnian
Hi,
You should remove some resistors.
Please refer to this link 
http://www.arducam.com/knowledge-base/raspberry-pi-multi-camera-adatper-board/?preview_id=1927&preview_nonce=1b5d1be442&preview=true

Regards.

hshbnian, 2017-08-07:
@UCTRONICS
Hello,
I'm not allowed to see this link ( I've received this error when I opened the link ==>" Sorry, you are not allowed to preview drafts.")

NCRbasketcase, 2017-09-20:
I can't get the link to work either

supprot, 2017-09-21:
@hshbnian
@NCRbasketcase
Hi,
Please refer to this link:
http://www.arducam.com/knowledge-base/raspberry-pi-multi-camera-adatper-board/

gaborsoter, 2018-01-07:
Hi,

What do you mean exactly by "use larger current power" or "shorter cable"?

Has anybody solved these issues mentioned before?

supprot, 2018-01-09:
@gaborsoter
About multi camera can't work fine,You should remove the resistance.
Please refer to this link:
http://www.arducam.com/knowledge-base/raspberry-pi-multi-camera-adatper-board/
About containing  so much noise, It has a lot to do with the length of the line and the current, so advice you use a shorter cable and the out current power should be 2.5A .you should ensure your power input.

garylzimmer, 2018-03-07:
Has the OP managed to find a solution to this?

ArduCAM, 2018-11-14:
@hshbnian   Would you please send us the hardware setup photo.  It looks like the cable length is too long which cause signal integrity issue. As you can see the camera CSI signal is very high speed.