ArduCAM/Arduino #246: Ov2640 set black and white and then set brightness.

wdouglas1999, 2017-08-04:
When I set the OV2640 to black and white it works.
When i set the brightness it works.
The issue is when I try to set them together the brightness works but the image is not black and white. 
Any way to fix this?

UCTRONICS, 2017-08-07:
@wdouglas1999
Hi,
You should check 2640 register configuration information. Maybe it exist conflict.

Regards,

Dzandaa, 2019-05-23:
I have the same problem with Brightness, Contrast, Saturation, etc...
The last command reset the other.

mosazahany, 2022-03-03:
Strange problems in adjusting the saturationand the brightness in ov2640
These bugs do not actually exist and are a problem of the available documents
When you execute the following commands
set_brightness(s,2);
set_contrast(s,2);
set_brightness(s,2);
The last command is always active and the previous commands are canceled.
To solve this problem, you must activate the desired effect and this is done with the following command

Run the following command at the end of all settings.

unsigned char enable_effect= 1(saturation and hue enable)+
2(aditional saturation enable)+
4(brighnes and contrast enable)+
8(spital effect1 enable green -> blue)+
16(spital effect1 enable gray -> read)+
32(remove (UV) in YUV color system)+
64(enable negative effect)+
128(remove (Y) in YUV color system) ;
write_SCCB(0xff, 0x00);
write_SCCB(0x7c, 0x00);
write_SCCB(0x7d,enable_effect);

command
write_SCCB(0x7c, ADRESS);
Preparing the address for data injection

ofter that use
The following command to inject data
write_SCCB(0x7d,byte_0);
write_SCCB(0x7d,byte_1);
write_SCCB(0x7d,byte_2);
.
.
.

The memory structure is as follows
byte_0 =enable_effect
byte_0->bit0 =saturation and hue
byte_0->bit1 =saturation enable
byte_0->bit2 =brighnes and contrast enable
byte_0->bit3 =green -> blue spitial effect (Antique and blunish and greenish and readdish and b&w) enable
byte_0->bit4 =gray -> read spitial effect (Antique and blunish and greenish and readdish and b&w) enable
byte_0->bit5 =remove (UV) in YUV color system
byte_0->bit6 =negative enable
byte_0->bit7 =remove (Y) in YUV color system
byte_1 =saturation1 0-255
byte_2 =hue 0-255
byte_3 =saturation2 0-255
byte_4 =reenter saturation2 in documents
byte_5 =spital effect green -> blue 0-255
byte_6 =spital effect gray -> read 0-255
byte_7 =contrast uper eadge 0-255
byte_8 =contrast lower eadge 0-255
byte_9 =brighnes 0-255
byte_10= if byte_10==4 contrast effective

[mosazahany@gmail.com](mailto:mosazahany@gmail.com)

با تشکر