ArduCAM/Arduino #249: Arducam mini 5MP with Arduino pro

Idavickit, 2017-08-09:
Hi @ArduCAM 

I'm a 5MP arducam mini camera, and trying to test with the "ArduCAM_Mini_Video_Streaming" example. 
But I got nothing in the host_app (ArduCAM_Host_V2) : no image. In the Serial monitor I could only see reverse question mark diplaying. Any help would be welcome, thanks. 
These are the connections I used : 
CS--> D10
MOSI-->D11
MISO-->D12
SCK-->D13
VCC-->5V
GND-->GND
SDA-->A4
SCL-->A5

UCTRONICS, 2017-08-09:
@Idavickit
Hi,
In our ArduCAM_Mini_Video_Streaming demo, we define CS 7.
Regards.

Idavickit, 2017-08-09:
Thank you for your answer @UCTRONICS 
I defined CS 10 instead of CS 7 since I'm using arduino pro board. I saw here [https://store.arduino.cc/usa/arduino-pro](url) that D10 is CS pin for this board. 

Although, I've just tried with CS 7 but it's still not working. 
I'm wondering if arducam is not compatible with this board...
Thanks

UCTRONICS, 2017-08-10:
@Idavickit
Hi,

Arducam should be compatible with this board.We have test it on Pro mini board.
You can test I2c and SPI separately.

Regards.