ArduCAM/Arduino #250: Transferring images over wifi with http POST

goetzmoritz, 2017-08-11:
Hi,

I'm trying to transmit the image data with http post but I have some issues.

This is my code for the esp:

```
void camCapture(ArduCAM myCAM){ 
  //WiFiClient client = server.client();
  WiFiClient client;
  const int httpPort = 80;
  if (!client.connect(host, httpPort)) {
        Serial.println("connection failed");
        return;
  }
  size_t len = myCAM.read_fifo_length();
  if (len >= 0x07ffff){
    Serial.println("Over size.");
    return;
  }else if (len == 0 ){
    Serial.println("Size is 0.");
    return;
  }
  
  myCAM.CS_LOW();
  myCAM.set_fifo_burst();
  #if !(defined (ARDUCAM_SHIELD_V2) && defined (OV2640_CAM))
  SPI.transfer(0xFF);
  #endif
  String response = "POST /api/device/image HTTP/1.1\r\n";
  response += "Content-Type: image/jpeg\r\n";
  response += "Content-Length: " + String(len) + "\r\n\r\n";
  
  Serial.println("conected to the server");
  client.println(response);
  
  static const size_t bufferSize = 4096;
  static uint8_t buffer[bufferSize] = {0xFF};
  
  while (len) {
      size_t will_copy = (len < bufferSize) ? len : bufferSize;
      SPI.transferBytes(&buffer[0], &buffer[0], will_copy);
      if (!client.connected()) break;
      client.write(&buffer[0], will_copy);
      len -= will_copy;
  }
  
  myCAM.CS_HIGH();
}
```

This is my node js code for receiving and saving the images:

```
http = require('http');
fs = require('fs');
server = http.createServer( function(req, res) {
    console.dir(req.param);
    if (req.method == 'POST') {
        var f = fs.createWriteStream('out.jpeg');
        req.on('data', function (data) {
            f.write(data);
        });
        req.on('end', function () {
            f.end();
        });
    }
});

port = 80;
host = '192.168.88.19';
server.listen(port, host);
console.log('Listening at http://' + host + ':' + port);
```

The images arrive corrupted:
- There is always a blank line at the beginning of the image-file
- The images arrive "crippled" (see sample-file attached).

I think that my header is not correct and that there are illegal characters in the jpeg but I'm not able to figure out what's wrong.

[image]

goetzmoritz, 2017-08-11:
Figured it out myself just right now.

In the header:

`response += "Content-Length: " + String(len) + "\r\n\r\n";`

should be:

`response += "Content-Length: " + String(len) + "\r\n";`

goetzmoritz, 2017-08-11:
Still one thing:
I'm not able to increase the resolution to maximum bei changing:

`myCAM.OV5642_set_JPEG_size(OV5640_320x240);`

to

`myCAM.OV5642_set_JPEG_size(OV5640_2592x1944);`

The complete code is:

```
#include <ESP8266WiFi.h>
#include <WiFiClient.h>
#include <ESP8266WebServer.h>
#include <ESP8266mDNS.h>
#include <Wire.h>
#include <ArduCAM.h>
#include <SPI.h>
#include <ESP8266HTTPClient.h>
#include "memorysaver.h"
#if !(defined ESP8266 )
#error Please select the ArduCAM ESP8266 UNO board in the Tools/Board
#endif

//This demo can only work on OV5640_MINI_5MP_PLUS or ARDUCAM_SHIELD_V2 platform.
#if !(defined (OV5640_MINI_5MP_PLUS)||(defined (ARDUCAM_SHIELD_V2) && defined (OV5640_CAM)))
#error Please select the hardware platform and camera module in the ../libraries/ArduCAM/memorysaver.h file
#endif

//host
const char* host = "192.168.88.19";
const int port = 80;

const int SHUTTER = 15;
int shutterState = 0;
// set GPIO16 as the slave select :
const int CS = 16;
//you can change the value of wifiType to select Station or AP mode.
//Default is AP mode.
int wifiType = 0; // 0:Station  1:AP

//Station mode you should put your ssid and password
const char* ssid = "xxx"; // Put your SSID here
const char* password = "xxxxx"; // Put your PASSWORD here

ESP8266WebServer server(80);

ArduCAM myCAM(OV5640, CS);

void start_capture(){
  myCAM.clear_fifo_flag();
  myCAM.start_capture();
}

void camCapture(ArduCAM myCAM){ 
  //WiFiClient client = server.client();
  WiFiClient client;
  const int httpPort = 80;
  if (!client.connect(host, httpPort)) {
        Serial.println("connection failed");
        return;
  }
  size_t len = myCAM.read_fifo_length();
  if (len >= 0x07ffff){
    Serial.println("Over size.");
    return;
  }else if (len == 0 ){
    Serial.println("Size is 0.");
    return;
  }
  
  myCAM.CS_LOW();
  myCAM.set_fifo_burst();
  #if !(defined (ARDUCAM_SHIELD_V2) && defined (OV2640_CAM))
  SPI.transfer(0xFF);
  #endif
  String response = "POST /api/device/image HTTP/1.1\r\n";
  response += "Content-Type: image/jpeg\r\n";
  response += "Content-Length: " + String(len) + "\r\n";
  
  Serial.println("conected to the server");
  client.println(response);
  
  static const size_t bufferSize = 4096;
  static uint8_t buffer[bufferSize] = {0xFF};
  
  while (len) {
      size_t will_copy = (len < bufferSize) ? len : bufferSize;
      SPI.transferBytes(&buffer[0], &buffer[0], will_copy);
      if (!client.connected()) break;
      client.write(&buffer[0], will_copy);
      len -= will_copy;
  }
  
  myCAM.CS_HIGH();
}

void serverCapture(){
  start_capture();
  Serial.println("CAM Capturing");

  int total_time = 0;
  total_time = millis();
  while (!myCAM.get_bit(ARDUCHIP_TRIG, CAP_DONE_MASK));
  total_time = millis() - total_time;
  Serial.print("capture total_time used (in miliseconds):");
  Serial.println(total_time, DEC);
  total_time = 0;
  Serial.println("CAM Capture Done!");
  total_time = millis();
  camCapture(myCAM);
  total_time = millis() - total_time;
  Serial.print("send total_time used (in miliseconds):");
  Serial.println(total_time, DEC);
  Serial.println("CAM send Done!");
}

void setup() {
  uint8_t vid, pid;
  uint8_t temp;
#if defined(__SAM3X8E__)
  Wire1.begin();
#else
  Wire.begin();
#endif
  Serial.begin(115200);
  Serial.println("ArduCAM Start!");

  // set the CS as an output:
  pinMode(CS, OUTPUT);

  // initialize SPI:
  SPI.begin();
  SPI.setFrequency(4000000); //4MHz

  //Check if the ArduCAM SPI bus is OK
  myCAM.write_reg(ARDUCHIP_TEST1, 0x55);
  temp = myCAM.read_reg(ARDUCHIP_TEST1);
  if (temp != 0x55){
    Serial.println("SPI1 interface Error!");
    while(1);
  }
  //Check if the camera module type is OV5640
  myCAM.wrSensorReg16_8(0xff, 0x01);
  myCAM.rdSensorReg16_8(OV5640_CHIPID_HIGH, &vid);
  myCAM.rdSensorReg16_8(OV5640_CHIPID_LOW, &pid);
   if((vid != 0x56) || (pid != 0x40)){
   Serial.println("Can't find OV5640 module!");
   while(1);
   }
   else
   Serial.println("OV5640 detected.");

  //Change to JPEG capture mode and initialize the OV5642 module
  myCAM.set_format(JPEG);
  myCAM.InitCAM();
  myCAM.write_reg(ARDUCHIP_TIM, VSYNC_LEVEL_MASK);   //VSYNC is active HIGH
  myCAM.OV5642_set_JPEG_size(OV5640_320x240);
  delay(1000);

  if (wifiType == 0){
    if(!strcmp(ssid,"SSID")){
       Serial.println("Please set your SSID");
       while(1);
    }
    if(!strcmp(password,"PASSWORD")){
       Serial.println("Please set your PASSWORD");
       while(1);
    }
    // Connect to WiFi network
    Serial.println();
    Serial.println();
    Serial.print("Connecting to ");
    Serial.println(ssid);
    
    WiFi.mode(WIFI_STA);
    WiFi.begin(ssid, password);
    while (WiFi.status() != WL_CONNECTED) {
      delay(500);
      Serial.print(".");
    }
    Serial.println("WiFi connected");
    Serial.println("");
    Serial.println(WiFi.localIP());
  }else if (wifiType == 1){
    Serial.println();
    Serial.println();

    Serial.println("");
    Serial.println(WiFi.softAPIP());
  }
  
  // Start the server
  server.on("/capture", HTTP_GET, serverCapture);
  server.begin();
  Serial.println("Server started");
}
void loop() {
  //server.handleClient();
  shutterState = digitalRead(SHUTTER);
  if (shutterState == HIGH) {
    // turn LED on:
    serverCapture();
    Serial.println("high");
    delay(1000);
  } else {
    // turn LED off:
    
  }
}
```

kotran88, 2018-04-20:
Did you write in host as 192.168.88.19 ???
because on my arduino IDE..it keep failed in connecting to my host.