ArduCAM/Arduino #253: Distorted videos - OV5642

AlexandreN84, 2017-08-16:
I'm capturing videos, and the image gets distorted [like this](http://i.imgur.com/Awt9IDo.jpg).

What should I do to remove these distortions?

I have an OV2640, and the images generated from it are sharper than the OV5642.

Hardware:		Arducam ESP8266 UNO V2
Module:  			Arducam ESP8266 UNO
Flash Size: 		4MB 
CPU Frequency:	80Mhz

### Sketch
https://github.com/ArduCAM/Arduino/tree/master/ArduCAM/examples/ESP8266/ArduCAM_ESP8266_OV5642_Capture

UCTRONICS, 2017-08-16:
@AlexandreN84
Hi,
Please try to refocuse.
Regards.

AlexandreN84, 2017-08-16:
@UCTRONICS  thanks for reply.
I did not understand your tip.
You mean for me to reorient the camera? How should I do this?

UCTRONICS, 2017-08-17:
@AlexandreN84
Hi,
Your picture looks very fuzzy, it should be the focal length is not good.
Please try tobturn the lens back to focus.

Regards.

AlexandreN84, 2017-08-18:
@UCTRONICS, worked, thank you!

closed 2017-08-18 by AlexandreN84