ArduCAM/Arduino #257: Problem testing kit

maligus74, 2017-08-31:
Hi there,

I just bought an arducam kit through amazon : https://www.amazon.com/gp/product/B012VSHK4M/ref=oh_aui_detailpage_o04_s00?ie=UTF8&psc=1

Including:
- ArduCam Shield V2
- ArduCam mini 2MP

I'm using an arduino Mega2560 R3>

I supposed that the camera is an OV2640 so i try couple example on arduino but without any success.
I'm wondering if my connection for the camera is good or no, the shield as 20 pins and I'm not really sure how to connect it.

I'm new student trying to learn arduino so if you have couple example who are "good" thanks to share them with me :)

maligus74, 2017-08-31:
UPDATE

I try to use the example ArduCAM_OV2640_Digital_Camera

I connect my Mini_2MP directly on the Mega2560 board (see pictures)

UCTRONICS, 2017-08-31:
@maligus74

Hi,
You should use those demos for Shield_V2 board
https://github.com/ArduCAM/Arduino/tree/master/ArduCAM/examples/Shield_V2
Please download new ArduCAM library from https://github.com/ArduCAM/Arduino.git.
Copy  the ArduCAM library and ArduCAM_Touch library and UTFT4ArduCAM_SPI library to the 
..\arduino\libraries path.
Then you can use the demos about Shield V2 board.those demos are under ..\Arduino\libraries\ArduCAM\example\Shield_V2 path.

Let us know if you need more help.
Regards.

maligus74, 2017-08-31:
I tried to put the new ArduCam library into the library folder. The OV7670 library is not valid so I just decide to remove this one and try the demo AdruCam_Mini_Capture2SD.
The program has been upload without any error but the problem now when I try to open ArduCam, I got the message: 
ArduCAM Start!

SPI interface Error!

i think I have a problem on the wiring. Actually my shield is directly connected to the Mega2560 and the camera plugged into the Arducam display. But the Arducam shield has 20 pins so I think i might be wrong in the connexion.

maligus74, 2017-08-31:
[image]
[image]

maligus74, 2017-08-31:
i tried different spot to connect the camera but nothing show up on arducam

maligus74, 2017-08-31:
I tried to change the const int SPI_CS = 7; replace by 10
The SPI is ok but now I get SD card error

maligus74, 2017-08-31:
New message now:
ArduCAM Start!

SPI interface OK.

SD Card detected.

Can't find OV2640 module!

UCTRONICS, 2017-08-31:
@maligus74
Hi,
Don't worry and we will try our best to help you.
Are you sure you are using OV2640 camera ?You can print the value of pid and vid.
 myCAM.wrSensorReg8_8(0xff, 0x01);
    myCAM.rdSensorReg8_8(OV2640_CHIPID_HIGH, &vid);
    myCAM.rdSensorReg8_8(OV2640_CHIPID_LOW, &pid);
    Serial.println(vid, HEX);
    Serial.println(pid, HEX);
If you are using OV2640 you will get pid = 0x26 and vid = 0x41 or 0x42.

What's more , The demos about Mini are used for Arducam mini camera instead of shield V2.
We advice you try ArduCAM_Shield_V2_Capture2SD demo and you should not change anything.

Let us know if you need more help.
Regards.

maligus74, 2017-08-31:
I tried with the shield but all time got the same fault:

#error Please select the hardware platform and camera module in the ../libraries/ArduCAM/memorysaver.h file

UCTRONICS, 2017-08-31:
@maligus74

Notice: You should enable the ARDUCAM_SHIELD_V2 platform and OV2640_CAM in the memorysaver.h
which is under the ..Arduino\libraries\ArduCAM path , for more problems, please refer to this link:
http://www.arducam.com/knowledge-base/arducam-typical-problems-arduino-uno/

maligus74, 2017-08-31:
Not sure now about the OV2640.
How could I check it ?

maligus74, 2017-08-31:
Found it on amazon on what I order:
Arducam Mini Module Camera Shield with OV2640 2 Megapixels Lens for Arduino UNO Mega2560 Board

maligus74, 2017-08-31:
[image]

maligus74, 2017-08-31:
I just correct the Step 2 and enable the #define OV2640_CAM

I don't have the error anymore but nothing show up yet on the display

maligus74, 2017-08-31:
I'm wondering if it's not a wiring issue.
Right now my camera (8 pins) is directly plugged in the arducam (20pins) but without to be sure where it suppose to go.

UCTRONICS, 2017-09-01:
@maligus74
Hi,
The Arducam mini 2mp (8pin) can't be used on Shield V2 board.
You should use this kind of camera:
http://www.arducam.com/camera-modules/2mp-ov2640/

Let us know if you need more help.
Regards.

maligus74, 2017-09-01:
Thanks for your answer. 

I still would like to try the camera (just to be more confident with). 
What's the best way to try ? Connect it directly to the Mega2560 ? If yes, what should be the wiring ?

UCTRONICS, 2017-09-01:
@maligus74
Hi,
If you want to connect the mini 2MP to 2560,
please refer to connection:

2640	VCC	GND	SDA	SCL	MISO	MOSI	SCK	CS
2560	5V	GND	 20	21	MISO	MOSI	SCK	10

Let us know if you need more help.
Regards.

maligus74, 2017-09-01:
Hi,

I tried the wiring like you said with the program ArduCam_Mini_Video2SD but the problem when I start the arducam software I got the message SD Card Error!.

Is there any way to not use the SD card ? Just look at the video on the arducam software ?

maligus74, 2017-09-01:
I comment the following line:
//Initialize SD Card
/*while(!SD.begin(SD_CS)){
  Serial.println(F("SD Card Error!"));delay(1000);
}*/
Now the arducam tell me 'OV2640 Detected' and then 'File Open fail'

UCTRONICS, 2017-09-01:
@maligus74
Hi,
Please send us your code and we will help you check the bugs.
If you want to send the image to the PC ,you can use ArduCAM_Mini_Video_Streaming
demo .
Notice: you should use the ArduCAM_Mini_QT software which is under ..\Arduino\libraries\ArduCAM\example\host_app\ArduCAM_Host_V2.0_Windows path.

Let us know if you need more help.
Regards.

maligus74, 2017-09-01:
Hi,

Thanks for all your answer.
I tried the program and it's running great !
I'm gonna buy this camera (https://www.amazon.com/gp/product/B013O8QB8O/ref=oh_aui_detailpage_o00_s00?ie=UTF8&psc=1) for the test with the shield, is that ok ?

If I get good result I will buy a better quality camera and play with that :)

UCTRONICS, 2017-09-03:
@maligus74
Hi,
Yes, This kind of camera can be pluged in the Arducam Shield board directly.
If you use this auto focus camera , please enable #define MT9D111B_CAM and #define ARDUCAM_SHIELD_V2
Let us know if you need more help.
Regards.

maligus74, 2017-09-05:
Hi,

I tried this week end to use the camera and the shield V2 but unfortunately it doesn't run.
Like you said I enable #define MT9D111B_CAM and #define ARDUCAM_SHIELD_V2.

I'm acutally trying with the ArduCam_Shield_V2_Video2SD but I have the fault #error please select the hardware platform and camera module...

Any idea ?

supprot, 2017-09-06:
@maligus74
Hi,Please try ArduCAM_Shield_V2_Digital_Camera demo or ArduCAM_Shield_V2_Camera_Playback demo.
Before uploading demo, please check those content in the code 
//This demo was made for Omnivision MT9D111A/MT9D111B/MT9M112/MT9V111_CAM/
//                                  MT9M001/MT9T112/MT9D112/OV7670/OV7675/
//                                  OV7725/OV2640/OV3640/OV5640/OV5642 sensor.
#if !(defined ARDUCAM_SHIELD_V2 && (defined MT9D111A_CAM|| defined MT9D111B_CAM || defined MT9M112_CAM  \
                                 || defined MT9V111_CAM || defined MT9M001_CAM || defined MT9T112_CAM \
                                 || defined MT9D112_CAM || defined OV7670_CAM  || defined OV7675_CAM  \
                                 || defined OV7725_CAM  || defined OV2640_CAM  || defined OV5640_CAM  \
                                 || defined OV5642_CAM|| defined OV3640_CAM))
#error Please select the hardware platform and camera module in the ../libraries/ArduCAM/memorysaver.h file
#endif

Let us know if you need more help.
Regards.

maligus74, 2017-09-06:
Hi,

I tried the code ArduCAM_Shield_V2_Digital_Camera and the part of the code that you gave me is correct.
Here is what I got on the display when I fired up: 
[image]

maligus74, 2017-09-06:
I got the same result with the ArduCAM_Shield_V2_Camera_Playback

maligus74, 2017-09-06:
Actually in the memorysaver.h I enable in Step 1 #define ARCUDAM_SHIELD_V2 and in the Step 2 #define MT9D111A_CAM (I also tried MT9D111B_CAM)

supprot, 2017-09-06:
@maligus74
Hi,
Don't worry and we  will try our best to help you solve this problem.Tomorrow we will test it
and give you a reply as soon as possible.

supprot, 2017-09-07:
@maligus74
Hi,
Very sory for our library's bug,wee have optimized it , please download our new library again from
https://github.com/ArduCAM/Arduino.git
Please enable the #define ARDUCAM_SHIELD_V2 and #define MT9D111B_CAM 
in the memorysaver.h file

Let us know if you need more help.
Regards.

maligus74, 2017-09-07:
Running great with the new library !
Thanks a lot :+1:

closed 2017-09-07 by maligus74

supprot, 2017-09-07:
@maligus74
Hi,
Glad to hear you have solved  problems.