ArduCAM/Arduino #259: SPI functions

goetzmoritz, 2017-09-06:
Hi there,

while it is nice that there are SPI-functions for the OV5642-Mini-Plus I wonder if it is possible to have those functions for OV5640-Mini-Plus, too? There are some available, but eg sharpness (edge enhancement in the datasheet) is not available yet as an SPI-function for OV5640. Another question: Do the EV-SPI-functions for OV5640 force fixed EV-values and disable automatic exposure for the OV5640

thanks!

supprot, 2017-09-06:
@goetzmoritz
Hi,
Please refer to http://www.uctronics.com/OV5640_App_Note.pdf
We have given all demos about the OV5640 functions .

goetzmoritz, 2017-09-06:
Thank you for your reply.

However: 

- Does the EV-values get fixed to a certain value or are they relative values like "measure the expose and add +3 EV"?
- Is there no way to reduce sharpness?
- Is there no way to reduce jpeg-compression-level?

Thank you!

supprot, 2017-09-06:
@goetzmoritz
Hi,
You can search the address of 0x4407 in the OV5640_App_Note.pdf , please try to reduce the value to get a higher quality jpeg picture.

goetzmoritz, 2017-09-06:
Thank you, that helped.
But there is no "easy" way to set the gain to fixed value (like ISO 100) and than play with shutter-speed, correct? It would be nice to have a ISP-functions like:
```
myCAM.OV5640_set_ISO(100);
myCAM.OV5640_set_shutterSpeed(50000); // in Microseconds, for 0.05 s
```

supprot, 2017-09-06:
@goetzmoritz
Hi,
You can refer to our ArduCAM library to encapsulate that functions .