ArduCAM/Arduino #261: Cant save images to SD card, blocky or corrupt images.

ctat, 2017-09-06:
Hello, 

I am using a Arduino Mega with the Arducam Mini-5MP-plus and am unable to save images to the SD card.  A correct image will arrive 1/100 of the time.  Other images are shown with scattered data or 0bytes and corrupt. I have attached an sample broken image.

When using the example code  and arducam.exe the camera sends the picture to the widows desktop app without issue, every time.

Do you have any possible solutions for this? 

108.jpg is a correct image, so in this case 108 captures taken before a successful image.

[image]

[image]
[image]

supprot, 2017-09-07:
@ctat
Hi,
We advice you use our adapter board without those long dupont wire which will be  instable.
Please try to use a short more dupont wire .

Let us know if you need more help.
Regards.

ctat, 2017-09-07:
Hello supprot,

Thankyou.  That has fixed my issue.  My question is why are the dupont wires unstable?

What is the maximum cable length the camera will allow?  Is it the SPI lines or the I2C that the dupont does not work well with?  I need to run the camera 500cm away from Arduino Mega.  Is it possible?

supprot, 2017-09-07:
@ctat

The longer wire will have larger iinterference for the data signal .
If you want to get remote transmission, we advice you use wifi .

Regards.

ctat, 2017-09-08:
@supprot - 

What is the maximum cable length that will work? is 30cm too long, what is the maximum length?
I want to place the camera outside of the enclosure box, about 30-50cm away from arduino.
Is this too long of length?

please let me know.

Thanks

supprot, 2017-09-09:
@ctat
Hi,
You can try to weld the dupont to the arducam directly instead of plug it.
30-50cm should be OK.