ArduCAM/Arduino #262: Arducam Mini upload to server via http POST utilising php script on server

ArduChecker, 2017-09-09:
Hi folks,
     I am desperately trying to get the following use case up'n'running, upto now without success.
     I think there is something wrong with the code staring with  _ myCAM.CS_LOW();_ in the 
    function _serverCapture_ ...it seems connection to server get lost...maybe an CS/SPI issue?
    Need really some advice.
    Many thx in advance!!
	 
	 **USE-CASE:**
	 After getting triggered by MQTT topic, a jpeg snapshot shall be taken and feeded to server via http POST where
	 a php script stores the data into defined file&directory
	 
	 **USED-HW:**
	 Arduino UNO
	 Ethernet Shield W5100 (hardcoded CS -> D10!)
	 Arducam Mini ov2640 ( used D7 as CS)
	
	 **USED-Library/IDE:**
	 https://github.com/ArduCAM/Arduino/tree/master/ArduCAM library
	 Arduino IDE 1.8.3
	 
	 **USED-Wiring:**
	 ArduCam   		CS   MOSI   MISO   SCK  GND	VCC SDA  SLC  	
	 Ethernet Shield      D7    D11      D12     D13   GND      5V    A4   A5
	 
	 **USED-Code:**
         on Arduino
 ` 
#include <Ethernet.h>
#include <SPI.h>
#include <Wire.h>
#include <PubSubClient.h>
#include <ArduCAM.h>
#include "memorysaver.h"

EthernetClient client;           
PubSubClient mqttClient(client);

IPAddress server(192, 168, 178, 63);  
ArduCAM myCAM(OV2640, 7);

byte mac[] = { 0xDA, 0xAD, 0xBE, 0xEF, 0xFE, 0xDD };

byte serverCapture( ArduCAM myCAM ){

  myCAM.clear_fifo_flag(); 
  myCAM.start_capture();
  
  while ( !myCAM.get_bit(ARDUCHIP_TRIG, CAP_DONE_MASK) );

  uint16_t length = myCAM.read_fifo_length();
  
  if ( (length >= MAX_FIFO_SIZE) || (length == 0) ) { return 1; }
  
  if ( !client.connected() ) { return 1; }
  else {  
    while( client.available() ) { String line = client.readStringUntil('\r'); } 
  } 
  
  String start_request = ""; 
  String end_request = "";
  String filename="snapshot.jpg";

  start_request =  start_request + "\n" + "--AaB03x" + "\n" + "Content-Disposition: form-data; name=\"datei\";filename="+filename+"\n" + "Content-Type: file" + "\n" + "Content-Transfer-Encoding: binary" + "\n" + "\n";
  end_request   = end_request + "\n" + "--AaB03x--" + "\n";
  
  uint16_t full_length = start_request.length() + length + end_request.length();

  client.println(F("POST /Upload.php HTTP/1.1"));
  client.println(F("Host: ")); client.println(server);
  client.println(F("Content-Type: multipart/related; boundary=AaB03x"));
  client.print(F("Content-Length: ")); client.println(full_length);
  client.print(start_request);

  myCAM.CS_LOW();
  myCAM.set_fifo_burst();
  SPI.transfer(0xFF); 

 uint8_t temp_last,temp = 0; 
 bool is_header = false;
 int i = 0;
 
 size_t bufferSize = 2048;
 uint8_t buffer[bufferSize] = {0xFF};

 while ( length-- ) {
      temp_last = temp;
      myCAM.CS_LOW();
      temp =  SPI.transfer(0x00);
      
   if ( (temp == 0xD9) && (temp_last == 0xFF) ) {
     buffer[i++] = temp;       
     myCAM.CS_HIGH();
     
     if ( !client.connected() ) { return 1;}
    
     client.write(&buffer[0], i);
     is_header = false;
     i = 0;
     break; 
   }  
     
   if ( is_header == true ) { 
     if (i < bufferSize) {
       buffer[i++] = temp;
     } else {
           myCAM.CS_HIGH();
           
           if ( !client.connected() ){ return 1; }
           
           client.write(&buffer[0], bufferSize);
           i = 0;
           buffer[i++] = temp;
       }        
   } else if ( (temp == 0xD8) & (temp_last == 0xFF) ) {
            is_header = true;
            buffer[i++] = temp_last;
            buffer[i++] = temp;   
     } 
 } 
     
 myCAM.CS_HIGH();
 client.println(end_request);
}   

   
void callback( char* topic, byte* payload, unsigned int length ) {
    if ( strcmp(topic,"snapshot" ) == 0) {
      if ( serverCapture(myCAM) == 1 ) {  Serial.println(F("Error!") ); } 
    }  
} 

void mqttReconnect() {
  while (!mqttClient.connected()) {
    if (mqttClient.connect("ArduCam")) {
      Serial.println(F("MQTT connected"));
      mqttClient.subscribe("snapshot");
    } else {
      Serial.print(F("MQTT connection failed, rc="));
      Serial.print(mqttClient.state());
      delay(2000);
    }
  }
}

void setup() {
 
  pinMode(7, OUTPUT);

  Wire.begin();
  Serial.begin(115200);
  SPI.begin();

  myCAM.write_reg(ARDUCHIP_TEST1, 0x55);
   
  if ( myCAM.read_reg(ARDUCHIP_TEST1) != 0x55 ) {
    Serial.println( F("SPI1 interface Error!") );
    while(1);
  }
  
   uint8_t vid, pid;
  
  myCAM.wrSensorReg8_8(0xff, 0x01);
  myCAM.rdSensorReg8_8(OV2640_CHIPID_HIGH, &vid);
  myCAM.rdSensorReg8_8(OV2640_CHIPID_LOW, &pid);

  if ( (vid != 0x26) || (pid != 0x42) ) {
    Serial.println( F("Can't find OV2640 module!") ); 
    while(1); 
  }
  
 myCAM.set_format(JPEG);
 myCAM.InitCAM();
 myCAM.OV2640_set_JPEG_size(OV2640_640x480);
 myCAM.clear_fifo_flag();

 Ethernet.begin(mac); 
 delay(1000);
  
 mqttClient.setServer(server, 1883);
 mqttClient.setCallback(callback);  
}

void loop() {  

  if (!mqttClient.connected()) {
    mqttReconnect();
  }

  mqttClient.loop(); 
   
}
 `
on Server
`
<?php

$upload_folder = 'upload/'; //Upload-directory

$filename = pathinfo($_FILES['datei']['name'], PATHINFO_FILENAME);
$extension = strtolower(pathinfo($_FILES['datei']['name'], PATHINFO_EXTENSION));

$allowed_extensions = array('jpg', 'jpeg');
if(!in_array($extension, $allowed_extensions)) {
 die("Invalid file extension. Only  jpg, und jpeg-files are allowed");
}

//Check file size
$max_size = 5000*1024; // 5 Mb
if($_FILES['datei']['size'] > $max_size) {
 die("Files above 5Mb not allowed");
}

//Check that picture does not have errors

if(function_exists('exif_imagetype')) { /
 $allowed_types = array(IMAGETYPE_PNG, IMAGETYPE_JPEG, IMAGETYPE_GIF);
 $detected_type = exif_imagetype($_FILES['datei']['tmp_name']);
 if(!in_array($detected_type, $allowed_types)) {
 die("Only Upload of pictures is allowed");
 }
}

$datum = mktime(date('H')+0, date('i'), date('s'), date('m'), date('d'), date('y'));

$filename = $filename.'-'.date('Y.m.d_H-i-s', $datum);

$new_path = $upload_folder.$filename.'.'.$extension;

if(file_exists($new_path)) { 
 $id = 1;
 do {
 $new_path = $upload_folder.$filename.'_'.$id.'.'.$extension;
 $id++;
 } while(file_exists($new_path));
}

move_uploaded_file($_FILES['datei']['tmp_name'], $new_path);
echo 'Successful upload: <a href="'.$new_path.'">'.$new_path.'</a>';
?>
`

ArduChecker, 2017-09-09:
...something went wrong with the formating...here comes the server php script again:

<?php

$upload_folder = 'upload/'; //Upload-directory
$filename = pathinfo($_FILES['datei']['name'], PATHINFO_FILENAME);
$extension = strtolower(pathinfo($_FILES['datei']['name'], PATHINFO_EXTENSION));

$allowed_extensions = array('jpg', 'jpeg');

if( !in_array($extension, $allowed_extensions) ) {
 die("Invalid file extension. Only  jpg, und jpeg-files are allowed");
}

//Check file size
$max_size = 5000*1024; // 5 Mb
if ( $_FILES['datei']['size'] > $max_size ) { die ( "Files above 5Mb not allowed" ); }

//Check that picture does not have errors

if ( function_exists('exif_imagetype') ) { 
 $allowed_types = array(IMAGETYPE_PNG, IMAGETYPE_JPEG, IMAGETYPE_GIF);
 $detected_type = exif_imagetype($_FILES['datei']['tmp_name']);
 if ( !in_array($detected_type, $allowed_types) ) { die ( "Only Upload of pictures is allowed" ); }
}

$datum = mktime(date('H')+0, date('i'), date('s'), date('m'), date('d'), date('y'));

$filename = $filename.'-'.date('Y.m.d_H-i-s', $datum);
$new_path = $upload_folder.$filename.'.'.$extension;

if ( file_exists($new_path ) ) { 
 $id = 1;
 do {
 $new_path = $upload_folder.$filename.'_'.$id.'.'.$extension;
 $id++;
 } while(file_exists($new_path));
}

move_uploaded_file($_FILES['datei']['tmp_name'], $new_path);
echo 'Successful upload: <a href="'.$new_path.'">'.$new_path.'</a>';
?>

martinberlin, 2018-12-06:
I made a simple bootstrap 4 PHP gallery, it has an upload example 
https://github.com/martinberlin/php-gallery
Preview here: 
https://fasarek.de/php-gallery/gallery/index.php those are photos uploaded using arducam and esp8266