ArduCAM/Arduino #264: How to change the resolution on BMP format

chougala, 2017-09-15:
Hello,
From https://github.com/ArduCAM/Arduino/blob/master/ArduCAM/ov2640_regs.h I see only one resolution option for BMP image,  OV2640_QVGA i.e. 320*240.

Can you please provide similar register address-data configurations for capturing different resolution images other than 320*240?

Thanks,
Chougala

supprot, 2017-09-17:
@chougala 
Hi,
We are also configuring more resolutions.We will release more demos as soon as possible.

chougala, 2017-09-18:
Hi,
Thanks for quick response.  Is it possible for you to point us to any
document which explains how to change the resolution.

Regards,
Girija

mjs513, 2019-08-23:
Hi UCTRONICS
I know this is an old post but I have a project that I need to change the BMP format to QQVGA (160x120).  I think I found the correct initialization code for this but not 100% sure it will work with the OV240 and Arducam.  Also what else would have to be changed in the code to default to the QQVGA format.  Attached is the code:
[qqvga.txt](https://github.com/ArduCAM/Arduino/files/3535573/qqvga.txt)

xiongyu0523, 2019-10-19:
@mjs513 thank you for your config. I'm also interesting in get 160x120 bitmap. Have you successfully run this config and get a good bitmap?

mjs513, 2019-10-19:
Unfortunately could never get it working properly.  Gave up and just did a image size in my code before I save the image.

xiongyu0523, 2019-10-20:
@mjs513 Yes, I also tried and failed. When you resize the image from QVGA to QQVGA, did you do it in memory after receive the whole image? I guess I might not have enough memory.