ArduCAM/Arduino #270: wemos d1 mini & OS5642

kuzmanzs, 2017-11-01:
Hi,

I have put together the two devices according the following schematic:
ArduCAM Camera ———- WeMos D1 mini 
============== ======================
CS    ————————-– D0 (GPIO 16)
MOSI  ————————-– D7 (GPIO 13)
MISO  ———————-—– D6 (GPIO 12)
SCK   ————————-– D5 (GPIO 14)
GND   ————————-- G (GND)
VCC   ————————— 3V3 (VCC)
SDA   ————————— D2 (GPIO 4)
SCL   ————————— D1 (GPIO 5)

I am using the ArduCAM_ESP8266_OV5642_Capture.ino from the examples folder.

I have uncommented the follwing lines in  the memorysaver.h : 
#define OV5642_MINI_5MP_PLUS
#define ARDUCAM_SHIELD_REVC	
#define OV5642_CAM

My camera module has a RevC label on the silk.
It creates a image but it looks like corrupt. Some parts can be recognised on the image at low resolution but it looks like bad. The situation is same in video mode.

Im running the camera module with 3V3.

Does someone any idea what is wrong? What is the difference between OV5642 and OV5642 Plus.

If I init with #define OV5642_MINI_5MP it failes on if (temp != 0x55) check.

kuzmanzs, 2017-11-02:
Arducam ESP8266 UNO board should be use.

Problem solved

closed 2017-11-02 by kuzmanzs

frogale, 2019-02-26:
I am trying to use 5642 5MP (Plus?) with Wemos D1 Mini as well.. This is what I found:
http://usemodj.com/2016/08/25/esp8266-arducam-5mp-ov5642-camera-wifi-video-streaming/
But I ended with running web server and corrupted JPEG output. I looked into files and had to remove first two bytes from JPEG file (00 00 FF D8 -> FF D8) but files are still corrupted. There must be another problem. Even with ArduCam examples I am not able to get any picture.

Did you throw up Wemos and used Uno board?

UCTRONICS, 2019-02-27:
@frogale 
I advise you to use Arducam ESP8266 UNO to test the camera and ensure the camera is normal.