ArduCAM/Arduino #271: Recording Speed to SD Card

ORANGEID, 2017-11-04:
Hi there

I was wondering if there any way to speed up the recording of images to SD card

I am using the arducam shield v2 with a Arduino UNO and the OV5642(2592x1944) currently it seems to take quite a while

If not possible to speed it up does switching to a raspberry pi help ? 

Any help would be greatly appreciated

supprot, 2017-11-05:
As you can see the Arducam SPI camera is designed for capturing still photos or short movie clips.
The bottle neck is the SPI interface speed and the speed of Arduino board. With faster MCU like STM32 will helps, because it has DMA channel for both SPI camera and the micro-sd card controller.

ORANGEID, 2017-11-06:
Thanks For getting back to me so quickly, the STM32 sounds interesting i don't suppose you have any documentation about how to use to it with the v2 shield ?

jsabby, 2017-11-07:
Dr. Jeffrey A Sabby
Associate Professor
Physics Department
Southern Illinois University Edwardsville

james-oab, 2017-11-28:
Hi There

So I bought my self a Maple mini and trying to get this work, but having a little problem with the verify the code, haven't got to trying to upload it yet

system: osX 10.11
example code: ArduCAM_Shield_V2_Digital_Camera.ino
hardware library: Arduino_STM32-master

this is the error message i am getting

In file included from /Users/----/Documents/Arduino/libraries/ArduCAM/ArduCAM.cpp:110:0:
/Users/----/Documents/Arduino/libraries/ArduCAM/ArduCAM.h:661:2: error: 'regtype' does not name a type
  regtype *P_CS;
  ^
/Users/----/Documents/Arduino/libraries/ArduCAM/ArduCAM.h:662:2: error: 'regsize' does not name a type
  regsize B_CS;
  ^
/Users/----/Documents/Arduino/libraries/ArduCAM/ArduCAM.cpp: In constructor 'ArduCAM::ArduCAM(byte, int)':
/Users----/Documents/Arduino/libraries/ArduCAM/ArduCAM.cpp:136:5: error: 'P_CS' was not declared in this scope
     P_CS  = portOutputRegister(digitalPinToPort(CS));
     ^
/Users/----/Documents/Arduino/libraries/ArduCAM/ArduCAM.cpp:137:5: error: 'B_CS' was not declared in this scope
     B_CS  = digitalPinToBitMask(CS);
     ^
/Users/----/Documents/Arduino/libraries/ArduCAM/ArduCAM.cpp:144:21: error: 'sbi' was not declared in this scope
       sbi(P_CS, B_CS);
                     ^

Is the any specific STM32 example for working with arducam ?

Any help would be greatly appreciated

Cheers

James

UCTRONICS, 2017-11-29:
@james-oab
Hi, 
We are very interested in your platform you are using.Can you tell us which platform your are using?
Where can we download the Arduino_STM32 library?
We hope work together with you.
Regards.
Arducam support team.

james-oab, 2017-11-29:
Hi

The library I am using can be download from here, https://github.com/rogerclarkmelbourne/Arduino_STM32

I also requested help on there surport forum 
http://www.stm32duino.com/viewtopic.php?f=2&t=2906
They have point out that this issue is down the way that code use flash access but i haven't chance to loo into this yet - but i fear it maybe beyond me so any help would be greatly appreciated

Cheers

James

UCTRONICS, 2017-11-29:
@james-oab 
Great, but what's your board platform? We plan to try it .

james-oab, 2017-11-29:
Sorry forgot to add that

I am using a maple mini, this is where i bought it from https://www.ebay.co.uk/itm/272238622756

Many Thanks

James

UCTRONICS, 2017-11-29:
@james-oab 
Hi,
This board has no one to continue to maintain, we advise you not to use it.
Regards.

james-oab, 2017-11-29:
Ok thats, I did try compiling the code for few different board that reported the same issue ? Is there a STM32 board you do recommend ?

Thanks

James

UCTRONICS, 2017-11-29:
@james-oab
Hi,

At present,we have not use STM32 on Arduino IDE develop environment .
If you want to use STM32, you can develop it on keil environment.

Regards,

james-oab, 2017-11-29:
Ok thanks for getting back to me, do you have a library for the keil or would it be starting from scratch ?

UCTRONICS, 2017-11-30:
@james-oab
Please give us your email address and we will give you our camera driver code.

james-oab, 2017-12-01:
Thanks my email is james.clarke03@gmail.com