ArduCAM/Arduino #272: Corrupted images with  ArduCAM mini ov5642 plus on Raspberry PI  SPI  problem?

CorlandoD, 2017-11-06:
[image]

Hello !

I have downloaded the Raspberry PI drivers for the  ArduCAM mini ov5642 plus and compiled them.

I have made it until the state in the picture , but when trying to open with an image viewer it doesn't seem to have any valid data , you can see the contents of the hex dump also.

I am thinking maybe it is the buffer , I mean the way we read from the buffer or the SPI bus ...something with it..... maybe the driver is buggy ..... 

This is the repository link : https://github.com/ArduCAM/RaspberryPi.git

Please advise on what to do ... I will play a little with the camera drivers , but for now , I hope I will find a quick solution from you .

Thanks !
Orlando
P.S ... I hope you can see the image.

supprot, 2017-11-13:
@CorlandoD
Hi,
Please refer to this link to download our new library.

https://github.com/ArduCAM/Arduino/tree/master/ArduCAM/examples/RaspberryPi

Please operate as follows:

https://github.com/ArduCAM/Arduino/blob/master/ArduCAM/examples/RaspberryPi/README.md

Let us know if you need more help.
Regards.

CorlandoD, 2018-12-13:
Thank you for the help. I will get back to the project soon. I will let you know if anything else gets in my way.