ArduCAM/Arduino #273: FIFO reigisters details , ArduCAM ov5642 mini plus 5mp

CorlandoD, 2017-11-06:
[image]

Hello !

I have an ArduCAM ov5642 mini plus 5mp camera which I would like to interface with an FPGA. 

I read the HW application note and there is no detailed description of the above registers (see picture) . Is there a document where they are explained or does someone have some practical knowledge related to these registers ?
By the way in the hw application note they refer to the sw section of the document for details on the regs  but I do not see any description of the regs , I do have the file called :OV5642_camera_new-module_software_application_notes_1.3_MESADA.pdf but I can not find anything

I am interested in the way JPEG images are stored in the FIFO and also what registers to consult in what manner to read from the FIFO. (I understand the way SPI works and extracts the data from the FIFO)

Please advise 
Orlando
Thanks !

martinberlin, 2018-12-05:
Bump. Also interested on this @UCTRONICS / @ArduCAM Mr. Lee please provide more details if they are available. Thanks in advance!

CorlandoD, 2018-12-13:
Hello Mr Lee when you have time please respond, or the ArduCAM team. 

Thank you very much !

ArduCAM, 2018-12-13:
These are the detail description of the registers. You can see how it works by reading the camera driver in our github.  https://github.com/ArduCAM/Arduino/blob/master/ArduCAM/ArduCAM.cpp 
The camera is mainly designed for Arduino, if you want to use on FPGA, you have to read the camera driver carefully how to access these registers via SPI interface.