ArduCAM/Arduino #274: ArduCAM2SD without ArduCAM shield

Achim68, 2017-11-07:
Hi, yesterday I got my ArduCAM Mini 2MP OV2640 and now wanted to connect it together with my SD-Card to my Arduino MEGA R3. I just want to take a picture one time per hour so I think I don't really need an ArduCAM shield, right? I wired it up like this:

Camera             Arduino           SD-Card 
CS           >        7               
MOSI      >        51        <         MOSI
MISO      >        50        <         MISO
SCK        >        52
GND       >        GND    <         GND
VCC        >         5V        <         VCC
SDA        >        20
SCL         >        21
                          9          <         CS

https://cdn.sparkfun.com/assets/8/f/f/6/5/50e5d529ce395f2f7a000000.png

What am I doing wrong? Please help me

supprot, 2017-11-13:
@Achim68
Hi,
Please refer to the below connection:

2640	VCC	GND	SDA	SCL	MISO	MOSI	SCK	CS SD_CS 
2560	5V	GND	20	21	MISO	MOSI	SCK	10     9
Regards.

UCTRONICS, 2018-07-05:
@93450101 
Yes of course. If you don't use our shield board, you should use a circuit board which can connect your SD card and the Arduino UNO. You can directly connect the Arducam Mini camera to Arduino UNO board.

93450101, 2018-07-05:
@UCTRONICS        WHICH PIN SHOUD I CONNECT SD CARD AND BOARD

93450101, 2018-07-06:
@UCTRONICS which pin can i use for sd card and arducam

UCTRONICS, 2018-07-06:
@93450101 
You can refer to this link
http://www.arducam.com/wp-content/uploads/2017/12/ArduinoConnection-.jpg

ArduCAM, 2018-07-06:
You can use any unused GPIO for SD card cs pin, the SPI pins are connected to the UNO SPI hardware pins which is also shared with Arducam camera.