ArduCAM/Arduino #276: Unable to make example work for arducam shield v2

Stryker69669, 2017-11-17:
I purchased this arducam: [](https://www.amazon.ca/Arducam-Shield-Camera-Arduino-Mega2560/dp/B012VSHK4M) and am trying to use the example ArduCAM_Shield_V2_Camera_Playback to learn how it works. My issue is that the sketch uploads but nothing occurs on the screen or with button presses. The serial monitor has outputted 3 different things: 
1. "SPI interface Error!" which I believe is caused by 
`temp = myCAM.read_reg(ARDUCHIP_TEST1);
  if (temp != 0x55)
  {
    Serial.println(F("SPI interface Error!"));
    delay(1000); continue;    
  } else{
    Serial.println(F("SPI interface OK"));break;
  } 	  ` (98-105) but I don't know what it requires me to do to fix it. I can't find documented anywhere regarding this issue.

2. "⸮⸮⸮⸮⸮⸮\⸮ܸ⸮\⸮⸮񦸇\⸮ܸ⸮⸮⸮⸮⸮\⸮⸮⸮⸮⸮⸮ ... etc". I haven't been able to find the source of this print but it occurs after the interface error

3. At one point I was able to make it output "background(69, 70, 70)" once about every second or so (the parameter values change frequently as well). I don't know how I managed it but the touchscreen and camera still remained unresponsive

I haven't been able to find anything that explains to me how to use the arducam or how to make the examples work. I followed all the instructions (like uncommenting proper devices in memorysaver.h) so I don't get why it doesn't work.

I'm hoping someone knows why this is happening as I've been researching this for hours to no avail. I appreciate any and all help. Thanks!

UCTRONICS, 2017-11-17:
@Stryker69669
Hi,
Please refer to this link to get morehelp:
http://www.arducam.com/knowledge-base/arducam-shield-v2-tutorial/

Stryker69669, 2017-11-17:
I've done all that properly. That's not where my issue is coming from.

UCTRONICS, 2017-11-18:
@Stryker69669
Hi,
Don't worry and we will try our best to help you.Have you test other demos for Shield V2 ?
You can attach us some pictures for your problems.