ArduCAM/Arduino #278: OV2640 DOA? need help

DigitalSocrates, 2017-11-21:
here is what i know

1. followed instructions on how to setup module with mega 2560 
http://www.arducam.com/how-to-connect-arducam-shield-to-mega-2560/#more-509 
cs - 7 and so on
2. used sample code for OV2640
3. SPI test passes but module is not recognized as vid and pid are 0
4. if i remove version test camera just seats in  and waits for capture to be done
while (!myCAM.get_bit(ARDUCHIP_TRIG, CAP_DONE_MASK)) {

Any recommendation ? or did i got a DOA module?

UCTRONICS, 2017-11-21:
@DigitalSocrates
Hi,
Have you tested your camera on Arduino UNO board?
Have you tested your I2c bus with oscilloscope?
Does the camera's I2c interface have a signal?
Regards.

DigitalSocrates, 2017-11-21:
No do not have Arduino NO board. 

AND No for oscilloscope - do not have one.