ArduCAM/Arduino #280: Host App doesn't display images from ov2640 mini connected directly to Uno board

rdwdfrst, 2017-11-26:
Hello,
I hope that you can help with this problem. I have connected my ov2640 mini to an arduino uno(I am using the uno first because it is most widely mentioned in examples). I modified the memsaver.h file to uncomment only the single ov2640 line. When I flash the uno with the ArduCAM_Mini_Video_Streaming sketch and watch the serial monitor in the arduino IDE I can see that it finds the camera module. Here is the output from the serial monitor(I added the last 2 comments to see how far the code was running):
ACK CMD ArduCAM Start!
ACK CMD SPI interface OK.
ACK CMD OV2640 detected.
ACK CMD InitCAM() completed.
ACK CMD Resolution set.

Now when I go to use the ArduCam Host V2 from the examples folder it seems to start ok, and connects to the camera when I press open: (msg "COM is open!")
If I press capture I get the message "ArduCAM Single" in the host app, and I see the size indicator in the host app increase as data is transferred but I never see an image. 
If I check the SaveImage box and then hit capture I get the same result (the size changes like it is transferring) and no file is created. If I change the resolution in the host app and then press capture the amount of data transferred changes accordingly(at 320x240 size =~3600, at 1600x1200 size=~85000), but still no images are displayed.
Any idea why the host app isn't displaying the images? Any help or suggestions would be appreciated.

Thanks.

UCTRONICS, 2017-11-26:
@rdwdfrst 
Hi,
Before writting image datas to Host V2, you should write "ACK IMG " firstly.
such as if ((temp == 0xD8) & (temp_last == 0xFF))
        {
          is_header = true;
          Serial.println(F("ACK IMG"));  // write it firstly!
          Serial.write(temp_last);
          Serial.write(temp);
        }
Please refer to our Arducam_MIni_Video_Streaming demo  line 360 from https://github.com/ArduCAM/Arduino/blob/master/ArduCAM/examples/mini/ArduCAM_Mini_Video_Streaming/ArduCAM_Mini_Video_Streaming.ino
Please try it again .

Let us know if you need more help.
Regards.

rdwdfrst, 2017-11-27:
Hello,
Thank you for your response.  
first: that is the sketch I am trying to run(ArduCAM_Mini_Video_Streaming from the examples directory) It does have the lines you indicated. (the only changes I made were to add a couple log lines to the serial port to see how far the code was executing. The last thing that seems to happen is the resolution change at line ~126. 

How many log messages should I see on a normal startup of the camera/arduino? 
Mine originally stopped after:
 ACK CMD OV2640 detected.

I added a couple more log messages(after line 122 and 124) to see if the 
InitCAM() or 
OV2640_set_JPEG_size(OV2640_320x240) 
statements were hanging. They seem to execute OK.  

Then, I also changed the line:(~line 124)
myCAM.OV2640_set_JPEG_size(OV2640_320x240) to 
 myCAM.OV2640_set_JPEG_size(OV2640_1024x768);
this seemed to work because the "size" indicator on the Host App shows a larger size after pressing the capture button in the host app. I just never get an image on the Host App, and when I watch the serial monitor on startup I never see anything in the startup messages past my:  "ACK CMD Resolution set." message.
can you suggest a next step?
thanks again.

p.s. 
I am using Windows 10 Pro version 1703 build 15063.729 in case that makes any difference.

UCTRONICS, 2017-11-28:
@rdwdfrst
Hi,
You can print the image data to serial port to check if it's header is 0xFF 0xD8 and  the end with 0xFF 0xD9.
You can add some prints after line 465:
 else if ((temp == 0xD8) & (temp_last == 0xFF))
    {
      is_header = true;
     Serial.println(F("ACK CMD Find the image header successfully!"));
      Serial.println(F("ACK IMG"));
      Serial.write(temp_last);
      Serial.write(temp);
}
If they can't exist  ,which means your image data exist errors.You should ensure your hardware connection is reliable.

Regards.