ArduCAM/Arduino #283: Image distortion with Feather Huzzah ESP8266 + OV5642 Mini 5MP

xelfer, 2017-12-19:
I have a Feather huzzah ESP8266 with the OV5642_MINI_5MP_PLUS (The only option that worked.. how do I tell if it's a "PLUS" or not? I purchased this: https://www.iot-store.com.au/products/arducam-mini-module-camera-shield-with-ov5642-5-megapixels-lens)

I've made the suggested changes to WiFiClient.h, ClientContext.h and lwipopts.h from https://github.com/ArduCAM/Arduino/issues/21

I've tried `static const size_t bufferSize = 2048;`

and I'm still getting results like this when I try to go any bigger than 320x240. Sometimes 640x480 works. Earlier when I was trying I managed to get 1600x1200 consistently without a problem but now I can't seem to get more than 1/4th of the photo. 

320x240 (working at the moment):
[image]

640x480 (working at the moment):
[image]

1024x768:
[image]

1280x960:
[image]

Full code I'm using is here: https://gist.github.com/xelfer/7707464220aff93390255fe97b033351

Any advice appreciated

UCTRONICS, 2017-12-19:
@xelfer
Hi,
What about the result on Arduino UNO platform?
You can test it on Arduino UNO platform.

xelfer, 2017-12-19:
I don't have any Arduino UNO's.. just the feather huzzah. And maybe a raspberry pi if I go digging in my cupboard..

I'm most annoyed as I had this working perfectly until I changed the wiring and plugged the camera straight into the breadboard.. but even when I reverted it back to how I started I still have the same issues :/

xelfer, 2017-12-19:
I also just moved it to where I wanted the camera to stay and even 320x240 doesn't work (slightly further from the wifi but by no means "Far away", it's within 10 metres of an access point. 

[image]

supprot, 2017-12-19:
Please avoid to use jumping wires, connec the camera with cable as short as possible, better with grounding.

xelfer, 2017-12-19:
> Please avoid to use jumping wires, connec the camera with cable as short as possible, better with grounding.

I'm just using the wires it came with and a breadboard, I can't solder it as it's a proof of concept at the moment. 

I did however just move the power jumpers right next to the camera power instead of the opposite end of the breadboard, like so:

[image]

and it actually seems to have improved my results! I just managed a 1600x1200 without error! higher didn't work but that's progress I guess. 👍 

[image]

supprot, 2017-12-19:
Good progress. Tidy up wires and try higher resolution.

xelfer, 2017-12-19:
Is there much more I can do on a breadboard without soldering?

ArduCAM, 2017-12-21:
You can try to connect the camera to Feather Huzzah board directly without the breadboard using short jumping wires.

xelfer, 2017-12-21:
My jumping wires seem even longer than the wires that came with it, but i'll see what I can do. Thanks for the advice all!

closed 2017-12-21 by xelfer