ArduCAM/Arduino #284: SPI1 interface Error! (OV2640 + ESP8266 Nano V2)

weston0718, 2017-12-22:
I am using ArduCAM (OV2640) library with a ESP8266 Nano V2 to run a capture example code (following the instruction, http://www.arducam.com/knowledge-base/esp8266-nano-v2-tutorial/), but it was not workable. 

ArduCAM (OV2640) library with a ESP8266 Nano V2: 
https://www.amazon.com/gp/product/B06VSX7KQC/ref=oh_aui_detailpage_o00_s00?ie=UTF8&psc=1

The wrong messgae is shown as below.  
-----------------------------------------------------------------------------------------------------------
ArduCAM Start!
SPI1 interface Error!

Soft WDT reset

ctx: cont 
sp: 3fff1980 end: 3fff1b90 offset: 01b0

>>>stack>>>
3fff1b30:  3fff0b34 3fff00a8 3fff0b34 40202584  
3fff1b40:  feefeffe feefeffe feefeffe feefeffe  
3fff1b50:  feefeffe feefeffe feefeffe feefeffe  
3fff1b60:  feefeffe feefeffe feefeffe 3fff0b60  
3fff1b70:  3fffdad0 00000000 3fff0b58 40207404  
3fff1b80:  feefeffe feefeffe 3fff0b70 40100718  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(1,6)

 ets Jan  8 2013,rst cause:4, boot mode:(1,6)

wdt reset
-----------------------------------------------------------------------------------------------------------

How can I fix the problem? Thank you.

UCTRONICS, 2017-12-24:
@weston0718 
Hi,
Don't worry, On ESP8266 Nano V2 platform, The SPI CS  is GPIO2.Please check your harware connection firstly.
Does your camera can work fine on other  platform ( such as arduino uno)?

weston0718, 2017-12-27:
@UCTRONICS 

Thanks for your reply, but I do not fully understand what you mean.
May you elaborate a bit more? Thanks. 
There is my hardware connection between ArduCAM-Mini and ESP8266 Nano V2. 

Arducam-Mini-2MP         ESP8266 Nano V2
              CS               --->         CS
              MOSI          --->         MOSI
              MISO          --->         MISO
              SCK            --->          SCK
              GND          --->          GND
              VCC           --->           VCC
              SDA           --->           SDA
              SCL            --->           SCL

izaoui, 2018-01-17:
I have a similar issue, could you please help me : 
As you can see in the picture below, I have a Wemos (  ESP8266 + MEGA ), but on my ESP, I do not have D7 (GPIO 13), nor D5 (GPIO 14) 
Please, help !

[image]

supprot, 2018-01-17:
@izaoui 
Hi,
Does your board has SPI interface and I2c interface?

izaoui, 2018-01-17:
Hi, thank you for your response, 
I have a panel named "ICSP" ( screen below, but I'm not sure this is what your are talking about )
Seems this panel is more from Mega, not from ESP8266
Let me know please, FYI, the card link below
https://www.banggood.com/fr/Wemos-Mega-WiFi-R3-ATmega2560ESP8266-32Mb-Memory-USB-TTL-CH340G-Compatible-For-Arduino-Mega-p-1205437.html?cur_warehouse=CN
[image]
Best regards, Iliès

izaoui, 2018-01-17:
Please, help ...

supprot, 2018-01-18:
@izaoui 
We don't have your board.
You should look up the schematic diagram of the board firstly.
Regards.

jbishop129, 2018-03-13:
@weston0718  @UCTRONICS  I recently bought this same board from Amazon, and am having the same problem. I did change my sketch to use GPIO2
```const int CS = 2;```
but still get the same error in the serial console as @weston0718 initially reported:
```
ArduCAM Start!
SPI1 interface Error!
```

Has anyone gotten this board working?

UCTRONICS, 2018-03-13:
@jbishop129 

We have tested  ArduCAM_ESP8266_Nano_V2_Capture demo with ESP8266 Nano V2 board.
We attached some pictures for you.
You can test your camera on Arduino UNO platform to check if the camera can work fine.
Can you attach us some pictures for your hardware connection?

[image]
[image]

jbishop129, 2018-03-13:
Thanks for the quick response. Please see below, using the "ArduCAM_ESP8266_OV2640_Capture" example. Also like to point out that in your screenshot, your example name is ArduCAM_ESP8266_**Nano**.... which makes me think it is a different or newer sketch. Can you please provide the complete sketch? 

I would like to wire up to a generic Arduino, however the sketch does not make it clear (to me anyway) the pin-out. So, I don't really have a place to begin.

[image]

UCTRONICS, 2018-03-13:
@jbishop129 

We have updated our ArduCAM ESP8266 UNO board. You should remove the old version and download the new version following http://www.arducam.com/knowledge-base/esp8266-nano-v2-tutorial/
You can preview the demo from https://github.com/ArduCAM/ArduCAM_ESP8266_UNO/blob/master/libraries/ArduCAM/examples/examples/ESP8266/ArduCAM_ESP8266_Nano_V2_Capture/ArduCAM_ESP8266_Nano_V2_Capture.ino

jbishop129, 2018-03-13:
Thanks @UCTRONICS , something seems to be flaky with either my board, or my Arduino IDE: I removed all my existing ArduCAM config, and started fresh with your instructions above, and uploaded the Nano_V2_Capture example sketch. To my surprise, I was able to access the video stream when connecting to it with my phone! First time! 

Next step, I wanted to connect it to my WiFi network, so I changed over to ```int wifiType = 0;``` and filled in my info under the ```//Station mode... ``` section. Uploaded the sketch again, and I'm back to square one with the ```SPI1 interface error!``` message. Then I tried again to upload the original example (after closing Arduino IDE and starting back up, disconnecting and reconnecting the board), and still with the error.

I'm going to try again tomorrow on a different PC with a fresh Arduino IDE install. I really want to get this board working!
I'll report back tomorrow, thanks for the help so far. Now to shift gears and get some dinner started :)

UCTRONICS, 2018-03-13:
@jbishop129 
OK,
Any problems,please feel free to contact me.
Maybe your hardware connection is defective ,you can try reconnect your camera to the nano V2 board.

Lvbin from UCTRONICS support team.

jbishop129, 2018-03-13:
I am at a loss on this one. I went ahead and installed a fresh Arduino IDE on a new PC, loaded up the board from the .json file, rebooted (just to be safe), loaded up the example "...._Nano_V2_Capture" sketch, and just as before, ```SPI1 interface error!``` 
Very frustrating, as I did get it to work once last night, but never again. By all accounts, this seems to be a defective board. I'll pursue replacement with the seller and try again. Will report back when I get a replacement... thank you for your assistance so far!