ArduCAM/Arduino #285: Error when using SD card with the ArduCAM 2MP (Capture2SD code)

Sunshine-Leona, 2018-01-02:
Hello and Happy New Year to you @ArduCAM ,

I have a problem with the following code: "ArduCAM_Mini_Capture2SD"

I am using the Arduino MEGA 2560; the original Arduino Wi-Fi Shield mounted on it (this one [https://www.arduino.cc/en/Guide/ArduinoWiFiShield)](url) + the ArduCAM Mini 2MP without the camera shield. I added a small change which is initializing Pin 53 as an output:

`
// ArduCAM demo (C)2017 Lee
// Web: http://www.ArduCAM.com
// This program is a demo of how to use most of the functions
// of the library with a supported camera modules, and can run on any Arduino platform.
//
// This demo was made for Omnivision 2MP/5MP sensor.
// It will run the ArduCAM 2MP/5MP as a real 2MP/5MP digital camera, provide both JPEG capture.
// The demo sketch will do the following tasks:
// 1. Set the sensor to JPEG mode.
// 2. Capture and buffer the image to FIFO every 5 seconds 
// 3. Store the image to Micro SD/TF card with JPEG format in sequential.
// 4. Resolution can be changed by myCAM.set_JPEG_size() function.
// This program requires the ArduCAM V4.0.0 (or later) library and ArduCAM 2MP/5MP shield
// and use Arduino IDE 1.6.8 compiler or above
#include <ArduCAM.h>
#include <Wire.h>
#include <SPI.h>
#include <SD.h>
#include "memorysaver.h"
//This demo can only work on OV2640_MINI_2MP or OV5642_MINI_5MP or OV5642_MINI_5MP_BIT_ROTATION_FIXED platform.
#if !(defined OV5642_MINI_5MP || defined OV5642_MINI_5MP_BIT_ROTATION_FIXED || defined OV2640_MINI_2MP || defined OV3640_MINI_3MP)
  #error Please select the hardware platform and camera module in the ../libraries/ArduCAM/memorysaver.h file
#endif
#define SD_CS 9
const int SPI_CS = 7;
#if defined (OV2640_MINI_2MP)
  ArduCAM myCAM( OV2640, SPI_CS );
#elif defined (OV3640_MINI_3MP)
  ArduCAM myCAM( OV3640, SPI_CS );
#else
  ArduCAM myCAM( OV5642, SPI_CS );
#endif

void myCAMSaveToSDFile(){
char str[8];
byte buf[256];
static int i = 0;
static int k = 0;
uint8_t temp = 0,temp_last=0;
uint32_t length = 0;
bool is_header = false;
File outFile;
//Flush the FIFO
myCAM.flush_fifo();
//Clear the capture done flag
myCAM.clear_fifo_flag();
//Start capture
myCAM.start_capture();
Serial.println(F("start Capture"));
while(!myCAM.get_bit(ARDUCHIP_TRIG , CAP_DONE_MASK));
Serial.println(F("Capture Done."));  
length = myCAM.read_fifo_length();
Serial.print(F("The fifo length is :"));
Serial.println(length, DEC);
if (length >= MAX_FIFO_SIZE) //384K
{
  Serial.println(F("Over size."));
  return ;
}
if (length == 0 ) //0 kb
{
  Serial.println(F("Size is 0."));
  return ;
}
//Construct a file name
k = k + 1;
itoa(k, str, 10);
strcat(str, ".jpg");
//Open the new file
outFile = SD.open(str, O_WRITE | O_CREAT | O_TRUNC);
if(!outFile){
  Serial.println(F("File open faild"));
  return;
}
myCAM.CS_LOW();
myCAM.set_fifo_burst();
while ( length-- )
{
  temp_last = temp;
  temp =  SPI.transfer(0x00);
  //Read JPEG data from FIFO
  if ( (temp == 0xD9) && (temp_last == 0xFF) ) //If find the end ,break while,
  {
    buf[i++] = temp;  //save the last  0XD9     
    //Write the remain bytes in the buffer
    myCAM.CS_HIGH();
    outFile.write(buf, i);    
    //Close the file
    outFile.close();
    Serial.println(F("Image save OK."));
    is_header = false;
    i = 0;
  }  
  if (is_header == true)
  { 
    //Write image data to buffer if not full
    if (i < 256)
    buf[i++] = temp;
    else
    {
      //Write 256 bytes image data to file
      myCAM.CS_HIGH();
      outFile.write(buf, 256);
      i = 0;
      buf[i++] = temp;
      myCAM.CS_LOW();
      myCAM.set_fifo_burst();
    }        
  }
  else if ((temp == 0xD8) & (temp_last == 0xFF))
  {
    is_header = true;
    buf[i++] = temp_last;
    buf[i++] = temp;   
  } 
} 
}

void setup(){
uint8_t vid, pid;
uint8_t temp;
Wire.begin();
Serial.begin(115200);
Serial.println(F("ArduCAM Start!"));
//set the CS as an output:
pinMode(SPI_CS,OUTPUT);
pinMode(53, OUTPUT):
// initialize SPI:
SPI.begin();
while(1){
  //Check if the ArduCAM SPI bus is OK
  myCAM.write_reg(ARDUCHIP_TEST1, 0x55);
  temp = myCAM.read_reg(ARDUCHIP_TEST1);
  
  if (temp != 0x55){
    Serial.println(F("SPI interface Error!"));
    delay(1000);continue;
  }else{
    Serial.println(F("SPI interface OK."));break;
  }
}
//Initialize SD Card
while(!SD.begin(SD_CS)){
  Serial.println(F("SD Card Error!"));delay(1000);
}
Serial.println(F("SD Card detected."));

#if defined (OV2640_MINI_2MP)
  while(1){
    //Check if the camera module type is OV2640
    myCAM.wrSensorReg8_8(0xff, 0x01);
    myCAM.rdSensorReg8_8(OV2640_CHIPID_HIGH, &vid);
    myCAM.rdSensorReg8_8(OV2640_CHIPID_LOW, &pid);
    if ((vid != 0x26 ) && (( pid != 0x41 ) || ( pid != 0x42 ))){
      Serial.println(F("Can't find OV2640 module!"));
      delay(1000);continue;
    }
    else{
      Serial.println(F("OV2640 detected."));break;
    } 
  }
#elif defined (OV3640_MINI_3MP)
  while(1){
    //Check if the camera module type is OV3640
    myCAM.rdSensorReg16_8(OV3640_CHIPID_HIGH, &vid);
    myCAM.rdSensorReg16_8(OV3640_CHIPID_LOW, &pid);
    if ((vid != 0x36) || (pid != 0x4C)){
      Serial.println(F("Can't find OV3640 module!"));
      delay(1000);continue; 
    }else{
      Serial.println(F("OV3640 detected."));break;    
    }
 } 
#else
  while(1){
    //Check if the camera module type is OV5642
    myCAM.wrSensorReg16_8(0xff, 0x01);
    myCAM.rdSensorReg16_8(OV5642_CHIPID_HIGH, &vid);
    myCAM.rdSensorReg16_8(OV5642_CHIPID_LOW, &pid);
    if((vid != 0x56) || (pid != 0x42)){
      Serial.println(F("Can't find OV5642 module!"));
      delay(1000);continue;
    }
    else{
      Serial.println(F("OV5642 detected."));break;
    } 
  }
#endif
myCAM.set_format(JPEG);
myCAM.InitCAM();
#if defined (OV2640_MINI_2MP)
  myCAM.OV2640_set_JPEG_size(OV2640_320x240);
#elif defined (OV3640_MINI_3MP)
  myCAM.OV3640_set_JPEG_size(OV3640_320x240);
#else
  myCAM.write_reg(ARDUCHIP_TIM, VSYNC_LEVEL_MASK);   //VSYNC is active HIGH
  myCAM.OV5642_set_JPEG_size(OV5642_320x240);
#endif
delay(1000);
}
void loop(){
myCAMSaveToSDFile();
delay(5000);
}

`

My connections are as follows:
MISO -> 50
MOSI -> 51
SCK -> 52
VCC -> 5V
SDA -> 20
SCL -> 21
Pin 53 of the Arduino board is connected to Pin 9 ( the SD_CS Pin). 

When I upload the code and check the serial monitor, I see reverse question marks. The ArduCAM host app just shows "ArduCAM Single Capture" when I open the port and click "Capture".

The "ArduCAM_Mini_2MP_OV2640_functions" code works fine with the same connections, excluding the SD pin, of course. I was hoping you would be able to point out my mistake. Thanks in advance!

supprot, 2018-01-03:
@Sunshine-Leona
Hi,
Can you detected your SD card?
On the Arduino WIFI Shield , the pin4 is used for SD cs.
You can try to change the pin 9 to pin 4.
Regards.

Sunshine-Leona, 2018-01-03:
@supprot 
Thank you for the fast  reply. I tried doing what you said. Here is the output of the serial monitor:

supprot, 2018-01-03:
@Sunshine-Leona 
Hi,
From your serial monitor, we can't see the state of SD card.
Have  you init your SD card?
//Initialize SD Card
while(!SD.begin(SD_CS)){
Serial.println(F("SD Card Error!"));delay(1000);
}
Serial.println(F("SD Card detected."));
the host app can't be used on "ArduCAM_Mini_Capture2SD" demo.
For "ArduCAM_Mini_2MP_OV2640_functions" demo,before opening the host app.You shoud close your serial monitor to prevent serial from being occupied.

Sunshine-Leona, 2018-01-03:
@supprot 
Hello again,

I finally got some pictures on the SD card.
But I am still getting the question marks. Why?
Serial Monitor Output:

ArduCAMrface OK.
SD Card detected.
OV2640 detected.
start Capture
Capture Done.
The fifo length is :6148
Image save OK.
0b⸮⸮2⸮⸮⸮⸮> >⸮H⸮0$⸮⸮⸮⸮݈I⸮⸮⸮⸮⸮⸮⸮⸮⸮ܘ!⸮⸮

supprot, 2018-01-03:
@Sunshine-Leona 

Hi, 

Do you see the picture in your SD card?
From the serial monitor,we think your SD card should work fine.
About the question marks, mybe because of  your wifi firmware. you can don't care it.

Let us know if you need more help.
Regards,

Sunshine-Leona, 2018-01-04:
Hello @supprot 

I figured out why I was getting the reverse question marks. It was due to the baud rate. I set it as "9600" and now I am getting a different problem. Look at the Serial Monitor Output below:

ArduCAM Start!
SPI interface OK.
SD Card detected.
OV2640 detected.
start Capture
Capture Done.
The fifo length is :7172
start Capture
Capture Done.
The fifo length is :8196
start Capture

It captures twice and just stops when it should be taking the third picture. "start Capture" and nothing after.

If I close and re-open the serial monitor, this is what I see:
ArduCAM Start!
SPI interface OK.
SD Card detected.
Can't find OV2640 module!
Can't find OV2640 module!
Can't find OV2640 module!
Can't find OV2640 module!
Can't find OV2640 module!
Can't find OV2640 module!
Can't find OV2640 module!
Can't find OV2640 module!

Also, something really weird happened to the SD card. I checked to see if the 2 captured pictures were saved onto the card. Instead, I found 0 KB pictures that were different from the ones I took. They were pictures taken by the ArduCAM from a previous date.

What could be the problem? Could it be the baud rate? Any help would be appreciated.

Thanks!

supprot, 2018-01-05:
@Sunshine-Leona 
Hi,
’Can't find OV2640 module!‘ means your camera hardware connection unstable.
If your picture is written to your SD card, you will see ‘Image save OK'.
We sugguest you check your camera to Arduino connection carefully.

Regards.
Arducam support team.

Sunshine-Leona, 2018-01-18:
@supprot 
Hello again,

As I have had trouble with the Capture2SD ArduCam example, I searched and found an alternative. 
This code was made by Vernel and it streams the JPEG bytes to the computer via Serial, which are then captured by Processing. The code they made was for the OV5642 ArduCAM. I edited the code for the OV2640 as it is the only one I have. Here is my code:
`
#include <Wire.h>

#include <ArduCAM.h>

#include <SPI.h>

#include "memorysaver.h"

//This code can only work on OV5642_MINI_5MP_PLUS platform.

#if !(defined (OV2640_MINI_2MP))

#endif

#define   FRAMES_NUM    0x00

// set pin 4 as the slave select for SPI:

const int CS1 = 4;

bool CAM1_EXIST = false;

bool stopMotion = false;

ArduCAM myCAM1(OV2640, CS1);

long int streamStartTime;

void setup() {

  // put your setup code here, to run once:

  uint8_t vid, pid;

  uint8_t temp;

  Wire.begin();

  Serial.begin(250000); //921600

  Serial.println(F("ArduCAM Start!"));

  // set the CS output:

  pinMode(CS1, OUTPUT);

  // initialize SPI:

  SPI.begin();

  //Check if the 4 ArduCAM Mini 5MP PLus Cameras' SPI bus is OK

  while (1) {

    myCAM1.write_reg(ARDUCHIP_TEST1, 0x55);

    temp = myCAM1.read_reg(ARDUCHIP_TEST1);

    if (temp != 0x55)

    {

      Serial.println(F("SPI1 interface Error!"));

[... 760 lines cut ...]
        if (CAM1_EXIST) {

          streamStartTime = millis();

          myCAMSendToSerial(myCAM1);

          double fps = ((millis() - streamStartTime) / 1000);

          Serial.println("Total Time: " + String(fps));

        }

        break;

      default:

        break;

    }

  }

}`

[https://github.com/Vernel/ArduCAM_OV562](url)

Sunshine-Leona, 2018-01-23:
@supprot 
Hello! I managed to get the code to work by changing what you told me and this part of the code:
myCAM1.rdSensorReg16_8(OV2640_CHIPID_HIGH, &vid);

myCAM1.rdSensorReg16_8(OV2640_CHIPID_LOW, &pid);

to the following:
myCAM1.rdSensorReg8_8(OV2640_CHIPID_HIGH, &vid);

myCAM1.rdSensorReg8_8(OV2640_CHIPID_LOW, &pid);

Thank you for all your help!

supprot, 2018-01-23:
@Sunshine-Leona
Hi,
Great to hear you have solved your problems.

pkvader, 2018-05-12:
I am having major difficulties with a similar issue. I have a SD breakout board and "Arducam Mini Module Camera Shield with OV2640 2 Megapixels Lens connected to my Arduino Mega board. The have it wired the same way and the app works with the first code in which I can take pictures and save to my computer. The problem is I am unable to save the pictures to the SD card. The second code for Capture2SD is not working for me and the app does not allow me to take photos it only states the COM is open. Please help!

UCTRONICS, 2018-05-12:
@pkvader 
The Capture2SD  demo can't support app, you should open your serial monitor and set the baud rate to 115200.

pkvader, 2018-05-12:
Is there a certain way to connect the SD card breakout board? The output serial monitor shows unknown characters only. I have on the SD breakout board's CS connected to PIN 53 and the Arducam's CS connected to PIN 10.

pkvader, 2018-05-12:
Update. It does work now Thank you for the quick responses. I appreciate the help!

UCTRONICS, 2018-05-12:
@pkvader 
Do you test it on arduino UNO board?
In the code , the SD CS is 9 and the camera CS is 7. You can try to change the CS pin

pkvader, 2018-05-12:
I used the Arduino Mega and for the SD breakout board i connected it to the ICSP pins the 2x4 matrix next to the reset button. It seems to be working but some of the files are corrupt still in the works but seems to have decent progress

pkvader, 2018-05-15:
Checked today and its not working again. getting "SD CARD ERROR" over and over again.

UCTRONICS, 2018-05-15:
@pkvader 
What SD card module do you use, most of which is the design problem of the module, which cause the
 The SPI bus was not released.

pkvader, 2018-05-15:
I am using Adafruit MicroSD card breakout board+ [ADA254].

pkvader, 2018-05-15:
when it bypasses the SD CARD ERROR issue I run into "File open failed" in the serial monitor

UCTRONICS, 2018-05-15:
@pkvader 

The The capacity of SD card is less than 8G.
You should format your SD card in fat 32

pkvader, 2018-05-15:
Nevermind I found out the problem I had. It was the breakout board being connected to the ICSP's 5V pin. It seems to be functioning better now. If it continues to work properly I'll make a short video highlighting some things not mentioned for the Arducam community.

pkvader, 2018-05-15:
Thank you again for the quick responses. I appreciate the time that you guys put in on this forum.

pkvader, 2018-07-01:
@KlydeLencioco 
Hey I currently do not have the project in hand but what is the issue that you are having. I can post the project in about a week when it is returned to me.

KlydeLencioco, 2018-07-02:
@pkvader we have the same issue. I sd card cant work with the arducam. I dont know if it is on my connection or other else.

KlydeLencioco, 2018-07-02:
@pkvader yes both of them work seperately. I am using arduino uni

pkvader, 2018-07-02:
It could be your wiring, are you using a breadboard + a lot of cable extensions? If so try to make the connections much shorter.

pkvader, 2018-07-02:
I used the sd breakout board by adafruit and the wiring needed to be pretty short otherwise it won’t work or work once in awhile. I used an arduino mega also not sure if you can use the uno but there are icsp pins on there so you can try it.

ArduCAM, 2018-07-02:
You have to be very carefully with the SD card read adapter, because it shares the SPI bus with Arducam. If the SD card occupy the SPI bus constantly, they can't work together. A lot of SD card read adapter boards are bad designed, there is no tristate control for the SPI bus at all.

KlydeLencioco, 2018-07-03:
I have also tried the icsp pins but it wont work also. 

@ArduCAM what is your suggestion for the sd card module that I need to use?

pkvader, 2018-07-03:
@KlydeLencioco 
What are you using exactly. List your project items. Plus what your goal is exactly. Mine was to only react to a Motion Detector to activate the ArduCam to take a picture then store that image onto an SD card through the AdaFruit SD breakout Board.

pkvader, 2018-07-03:
@KlydeLencioco 
Also your code if you modified it and/or which sample code are u using from the ArduCam library.

KlydeLencioco, 2018-07-04:
@pkvader 

My components are:
Micro sd module
ArduCam ov5642 5mp plus
arduino uno
pir motion detector
esp8266 wifi module

My project is the same as yours. The arducam need to capture an image when motion is detected and save it to the sd card. Right now, I did not hook up the esp8266 module because I have problem saving the image to the sd card.

I am using the ArduCam_Mini_5MP_Plus_Multi_Capture2SD program. I did not modify the code and used the pins for the cam cs and sd cs with respect to the ones declared in the program.

UCTRONICS, 2018-07-04:
@KlydeLencioco 
If you are using the ESP8266 module, please use our ArduCAM_ESP8266_UNO_V2_Capture2SD demo.

93450101, 2018-07-05:
@ArduCAM hello . I want to collect image data and rebuild image by data i recived ,.can you help me???????????????

KlydeLencioco, 2018-07-09:
@UCTRONICS do your shield with micro sd slot works perfectly even with the atmega328p chip?

UCTRONICS, 2018-07-09:
@KlydeLencioco 
Yes, our Shield V2 board is fully compatible with UNO board. For the SD card, we used the pin 9 as the CS of the SD card and share the hardware SPI with the camera.

KlydeLencioco, 2018-07-09:
@UCTRONICS where can I place my order?

ArduCAM, 2018-07-09:
Our micro SD card adapter is well designed to work with ATMEGA328p, you can find it from: 
http://www.uctronics.com/arducam-mini-multi-camera-adapter-board-for-arduino-raspberry-pi.html

KlydeLencioco, 2018-07-09:
does it need an external power supply?

ArduCAM, 2018-07-09:
If running one camera you don't need external power supply, If you want to use 4 cameras you might need extra micro-usb cable to the phone charger.

KlydeLencioco, 2018-07-10:
@ArduCAM will that shield also works if I have an esp8266 module?

pkvader, 2018-07-10:
[image]

pkvader, 2018-07-10:
[image]
[image]

pkvader, 2018-07-10:
@KlydeLencioco

pkvader, 2018-07-10:
I couldn't find the arducam online for the fritzing diagram but you get the point.  If you have it wired like this properly, its your code that could be the problem. The SD card readers usually require a stable connection meaning very short wires that connect it to the Arduino. If the wire is extended the data transfer will not work. @KlydeLencioco

93450101, 2018-07-13:
@supprot hello ,I want to display the image inside the SD card with the host app، can you help me?