ArduCAM/Arduino #292: inverted image (how do I fix), "stuck" framerate

coldstart01, 2018-01-25:
Hi Lee, first time commenting here. I have the Arducam MT9V034_VGA & USB sheild. Please help! Somewhere in the 2 days that I have had this the image became inverted. I don't know how to reset the sensor. Also the FPS never changes: I can set register 11 to 3 or 0x20 or 0x40 and its always reading back a framerate of 37 - advertised of course at 60.

coldstart01, 2018-01-25:
Addendum:
I switched from the USB 3.0 I was using and restarted with USB 2.0. My FR went immediately to 62-63 FPS both capture and exposure. However there was not a significant difference in frame rate between the 3 configurations available to me:

MT9V034_VGA_8bit
MT9V034_WVGA_8bit
MT9V034_WVGA_10bit

Also, do the registers stay set when not using the sensor?

One big problem persists: my image is inverted. How do I fix?

ArduCAM, 2018-01-26:
Please change the register 0x0D bit[5]=Column Flip, so the correct register setting is REG = 0x0D, 0x320 in order to mirror the image. 
These three settings are the same resolution 640x480 or 752x480 so their frame rate is the almost the same.
Let me know what do you want to do with the frame rate. You can reduce the resolution to get higher fps.

coldstart01, 2018-01-26:
Thanks! Those register settings did it. No longer mirrored.

I would like to have one more frame rate config. around 320 x 240 or slightly larger for testing. 

Also is here some configuration with the lens or filters for NIR/IR operation?

ArduCAM, 2018-01-26:
You have to change the width and height registers register for 320x240 resolution, I will work on that settings to then get back to you. 
I'm not sure what kind of camera you got and where it got from? We have lots of lens options http://www.arducam.com/lenses/
For NIR you just simple remove the IR cut filter on the bottom of the lens.

coldstart01, 2018-01-26:
Default M12 mount 6mm LS-6020 lens
1 pcs Arducam CMOS MT9V034 1/3-Inch 0.36MP Monochrome Camera Module
ArduCAM USB Camera Shield

coldstart01, 2018-02-15:
Lee.

Hi, did you ever get the change in resolution working (above)

coldstart01, 2018-05-08:
Have been working on other aspects of the project. I tried your registry setting above REG = 0x0D, 0x320, it works, but I have to do it every time I boot up the camera. 
1. is there a way to make it permanent?
2. what registry read/write vals to change resolution to 320 x 240 for example.

ArduCAM, 2018-05-09:
If you are using windows, you can write the regsiter settings into the MT9V034.cfg file permanently.
To change to 320x240, modify the following registers:
REG = 0x03, 240
REG = 0x04, 320

coldstart01, 2018-05-10:
Its MT9V022, linux thx is the same?

ArduCAM, 2018-05-11:
For linux, you have to set the regsiter in the code like this:
    ArduCam_writeSensorReg( useHandle, 0x03,  0x01E0 );
    ArduCam_writeSensorReg( useHandle, 0x04,  0x02F0 );
MT9V022 is almost the same as MT9V034.

coldstart01, 2018-05-18:
Im trying the linux samples and am getting library errors...do you know what the problem might be?
...
make all 
Building target: MT9V022
Invoking: GCC C++ Linker
g++ -L/home/primeuser/Documents/Marble/MT9V022/src -o "MT9V022"  ./src/MT9V022_demo.o   -lArduCamLib.so -lusb-1.0.so
/usr/bin/x86_64-linux-gnu-ld: cannot find -lArduCamLib.so
/usr/bin/x86_64-linux-gnu-ld: cannot find -lusb-1.0.so
collect2: error: ld returned 1 exit status
makefile:45: recipe for target 'MT9V022' failed
make: *** [MT9V022] Error 1

ArduCAM, 2018-05-19:
You also need the following library files for compiling your code.
https://github.com/ArduCAM/ArduCAM_USB_Camera_Shield/blob/master/Linux_x86/Cpp/ArduCAM_Linux_SDK_OpenCV_MT9V022_Demo/libArduCamLib.a 
https://github.com/ArduCAM/ArduCAM_USB_Camera_Shield/blob/master/Linux_x86/Cpp/ArduCAM_Linux_SDK_OpenCV_MT9V022_Demo/libusb-1.0.a

coldstart01, 2018-05-19:
Ok. Do they go in the usr/local/lib or just the /lib

coldstart01, 2018-05-20:
I just want a working copy.
`./Test
./Test: error while loading shared libraries: libopencv_core.so.2.4: cannot open shared object file: No such file or directory
`
So **_recompiled_** with opencv linking to latest version:
g++ MT9V034_demo.cpp -o Test `pkg-config --cflags --libs opencv` -lArduCamLib -lpthread -lusb-1.0 -L. -I. -std=gnu++11
I've added .a to my latest files. When I **recompile** to link to present opencv version I get error:
`/usr/bin/x86_64-linux-gnu-ld:/usr/local/lib/libArduCamLib.so: file format not recognized; treating as linker script
/usr/bin/x86_64-linux-gnu-ld:/usr/local/lib/libArduCamLib.so:0: syntax error
collect2: error: ld returned 1 exit status
`
If I could just get this running won't have t bother you guys : )

coldstart01, 2018-05-23:
Ok, so I updated the libraries and recompiled from within the demo code and it compiled successfully, no linker errors. Now, the window opens but shows an image very rarely; it sas "cam_data_len" error or something like that. I've tried redompiling, etc., same message.

coldstart01, 2018-05-25:
Sorry late
[image]

coldstart01, 2018-05-25:
Here is another error I picked up: failed to load canberra-gtk-module
[image]

coldstart01, 2018-05-25:
compiler statement:
g++ MT9V022_demo.cpp -o Test `pkg-config --cflags --libs opencv` -lArduCamLib -lusb-1.0 -lpthread -L. -I. -std=gnu++11

from within MT9V022 demo folder
**Demo folder:**
ArduCamLib.h	    libArduCamLib.so.1.0    libusb-1.0.so.0	 Test
libArduCamLib.a     libArduCamLib.so.1.0.0  libusb-1.0.so.0.1.0  Test-0
libArduCamLib.so    libusb-1.0.a	    libusb.h		 Test-1
libArduCamLib.so.1  libusb-1.0.so	    MT9V022_demo.cpp

coldstart01, 2018-05-25:
I'm not exactly sure what you mean. Can you plz clarify?
FYI the 022 and 034 both work excellently in Windows USB Demo. As you have pointed out it isnt yet ready for opencv.

The Linux based will play a few frames above error until crash on the Virtual Machine. I've tried different USB configs...my keyboard and mouse are USB. We need to make this app work. Does the USB driver need an update?

ArduCAM, 2018-05-26:
Please don't use Virtual machine, the performance of USB is very low in VM.

coldstart01, 2018-05-26:
What would you suggest - a dual boot OS?

ArduCAM, 2018-05-28:
Yes, you need to install dual boot OS for both Winodws and Ubuntu Linux

coldstart01, 2018-05-28:
I've got (2) machines. Can I dedicate 1 to each OS?

ArduCAM, 2018-05-28:
It will be even better, you can swap the camera if you come across any issue to do some comparison.

coldstart01, 2018-06-05:
Hi Mr.Lee, have 2 separate systems now 1 ubuntu and 1 windows. The opencv installation went well in linux/ubuntu. The program re-compiles fine to link to more modern opencv version with no linker errors.
So why does it start the program without following through:

`ArduCAM_Linux_SDK_Opencv_MT9V022_Demo# sudo ./Test
Device found = 1 
ArduCam_open successful
create capture thread successfully
create display thread successfully
`
That's all that happens.
No Window/image generation...

coldstart01, 2018-06-05:
By the way this is my compiler line from command prompt:
<code>
g++ MT9V034_demo.cpp -o Test `pkg-config --cflags --libs opencv` -lArduCamLib -lpthread -lusb-1.0 -L. -I. -std=gnu++11
</code>

coldstart01, 2018-06-09:
Keep trying and get this:

sudo ./Test
Device found = 1 
ArduCam_open successful
create capture thread successfully
create display thread successfully
Nothing else

ArduCAM, 2018-06-09:
please leave your email address, I will send you a new code to try it out.

coldstart01, 2018-06-09:
clangray@fastmail.com

coldstart01, 2018-06-10:
root@primeuser-MS-7760:~/Desktop/ArduCAM_Linux_SDK_OpenCV_MT9V022_Demo# sudo ./Test
<code>
Device found = 1 
shipAddr 0x46 value is: 5
Sensor 0x03 value is: 1e0
Sensor 0x04 value is: 2f0
Sensor 0x0B value is: 200
ArduCam_open successful
create capture thread successfully
create display thread successfully
USB_CAMERA_DATA_LEN_ERROR
USB_CAMERA_DATA_LEN_ERROR
USB_CAMERA_DATA_LEN_ERROR
USB_CAMERA_DATA_LEN_ERROR
USB_CAMERA_DATA_LEN_ERROR
USB_CAMERA_DATA_LEN_ERROR
USB_CAMERA_DATA_LEN_ERROR
USB_CAMERA_DATA_LEN_ERROR
USB_CAMERA_DATA_LEN_ERROR
USB_CAMERA_DATA_LEN_ERROR
USB_CAMERA_DATA_LEN_ERROR
USB_CAMERA_DATA_LEN_ERROR
USB_CAMERA_DATA_LEN_ERROR
USB_CAMERA_DATA_LEN_ERROR
...inifinty
</code>

ArduCAM, 2018-06-10:
[image]
It seems that your received data is not correct size or encounter the dropping frame issue.
I run the same code without any issue. Can you show me your hardware setup?

coldstart01, 2018-06-10:
I've attached the hardware setup report. Basically Ubuntu 18 on 250GB SSD drive, NVidia 770 card, some external USB hard drives, USB mouse and keyboard. Its all in the report, I hope its not too much information.
[hardinfo_report.txt](https://github.com/ArduCAM/Arduino/files/2088222/hardinfo_report.txt)

ArduCAM, 2018-06-10:
You'd better not to connect both the camera and USB hard drivers in the same USB root hub, they will race the bandwidth. Can you send me a photo of your camera board? I have to determine the part number.

coldstart01, 2018-06-10:
[image]
[image]

Can disassemble if u need

ArduCAM, 2018-06-10:
Thanks for your photos. 
It is really strage if it is working under Windows with the same hardware setup, but stop working under linux.

coldstart01, 2018-06-11:
NP.

Actually the Windows installation is now installed all by itself on a separate dedicated machine(#1). The second original machine (#2) is reinstalled with dedicated Ubuntu, erasing the VMWare. However it seems clear that the USB problem in (#2) is persisting from VMWare in the dedicated Ubuntu installation.

Tell me what you want to do. The next logical step to me would be to convert the Windows machine to dual-boot with Ubuntu and drop the current USB situation on the Ubuntu all together. It will be a pain, but I've got to get this camera going. Do you have a better idea?

coldstart01, 2018-06-13:
Also have you been able to get these cams to work with Linux?

ArduCAM, 2018-06-14:
As you can see from my previous message, I send you the screen shot for my result.
can you install the dual-boot OS on the same machine?

coldstart01, 2018-06-14:
Hi 
Which machine do you want me to add dual boot? My windows machine or my ubuntu machine. I assume you are talking about dual boot by adding ubuntu to my windows.

coldstart01, 2018-06-16:
Hi Lee how do you want me to configure the boot change? I assume its adding the Linux Ubuntu to the windows machine.

coldstart01, 2018-06-20:
Hi Lee, should I go ahead and add dual boot system to my windows?

ArduCAM, 2018-06-20:
We will release a new USB2.0 camera with onboard frame buffer, so the dropping frame issue will be fixed with that one.

coldstart01, 2018-06-20:
OK, when do you expect it to be released for purchase? OpenCV ready? Also, what I really like about the the 034 and the 022 is the global shutter. It is essential. Are you all going to keep that?

ArduCAM, 2018-06-20:
Actually we are ready to ship the new USB2.0 camera with onboard frame buffer. It will support more cameras including the MT9V022/034, AR0134 global shutter cameras.

coldstart01, 2018-06-20:
Great hit me with the link
Do you recommend still to do dual or try on mu Ubuntu first w/new cam

ArduCAM, 2018-06-20:
The listing isn't udpate yet: http://www.uctronics.com/arducam-usb-camera-shield.html but we will ship the new ones.

coldstart01, 2018-06-20:
Ok, need some assurances when I receive it is that it is the newest
module incorporating the frame buffer changes. People I work for.
--
  Gray Family
  clangray@fastmail.com

Links:

  1. https://github.com/ArduCAM/Arduino/issues/292#issuecomment-390405367
  2. https://github.com/

coldstart01, 2018-06-20:
BTW< between 022, 034 and 134 which is the  best? I need global shutter...fast fr

ArduCAM, 2018-06-20:
Please leave us a note after your order. MT9V022 we have is 0.36MP mono, MT9V034 is color at 60fps, and AR0134 is 1.2MP at 45fps.

coldstart01, 2018-06-20:
MT9V034 is color 
??mine was monochrome

ArduCAM, 2018-06-20:
Now we only have color MT9V034, and mono MT9V022. The MT9V022 is almost the same as MT9V034

coldstart01, 2018-06-21:
ArduCAM USB3.0 Camera Shield
...this is new...I should still get the 2.0 with frame buffer though right

ArduCAM, 2018-06-21:
USB3.0 doesn't have onboard frame buffer, not suitable for embedded system like Raspberry pi.

coldstart01, 2018-06-21:
Sorry to ask so many questions  : )
My platform is both PC and Linux separate, not rasperry PI. 

Links:

  1. https://github.com/ArduCAM/Arduino/issues/292#issuecomment-399128592
  2. https://github.com/

ArduCAM, 2018-06-21:
USB3 now only supports Windows, we are working on linux SDK now.

coldstart01, 2018-06-21:
Hi, I've placed an order based on our conversation. Ive sent a note to sales at utronics requesting to be sure to fill USB 2.0 with framebuffer. Can you please oversee that order and make sure its shipped correctly?

ArduCAM, 2018-06-21:
Thank you. We don't have old version at all. You won't miss it.

coldstart01, 2018-06-28:
Hi,
I got the new camera in the mail today. Please refer me to the procedure in linux:

-compile line & files
-driver!! : ) do I need one in Ubuntu

Thx!

ArduCAM, 2018-06-28:
The driver is compatible with old ones in this github: https://github.com/ArduCAM/ArduCAM_USB_Camera_Shield/tree/master/Linux_x86

coldstart01, 2018-06-28:
ok. Embarrassed to ask: how do I install a driver in linux/ubuntu? I did fine with windows install and deativating driver signature requiremen, no prob.

**How to install driver in linux.** Its a new linux dedicated system.

[3 more replies]