ArduCAM/Arduino #295: Arducam mini OV2640 Not Working - Won't Take Image

jasonc2010, 2018-02-04:
I recently purchased an ArduCAM OV2640 Camera Shield for Arduino and cannot get it to take an image. I guarantee it is wired correctly (I have tried CS pin on 7 and 10 by changing it in the code and wiring respectively), I have used the ArduCAM_Mini_2MP_OV2640_functions, ArduCAM_Mini_LowPowerMode and ArduCAM_Mini_Video_Streaming example code and have tried 2 different Sparkfun Redboards (Arduino Uno clones) and 1 Arduino Pro Mini. In the Serial monitor from every board I am getting the following messages: 

"ACK CMD ArduCAM Start!
ACK CMD SPI interface OK.
ACK CMD OV2640 detected."

I have wired the camera to both the header pins and the ISP pins with the same outcome. When using the ArduCAM_Host_V2 and clicking "OPEN" it will state that the "COM is open!" (see attached image) but never states anything else. NO error messages and I have NEVER seen the "ArduCAM Start!", "SPI interface OK!" or "OV2640 detected." messages like in the tutorial https://www.youtube.com/watch?v=hybQpjwJ4aA&feature=youtu.be. Similar results are found using the ArduCAM_host_v1.0. At this point I am assuming that the camera module is defective but I'd like to know if there is something I can try before returning it. The chips on both sides of the camera circuit board are noticeably warm to the touch but not hot enough to burn so I'm assuming this is normal? 

Thank you
[image]

supprot, 2018-02-04:
@jasonc2010
Hi,
Don't worry, The ArduCAM_Host_v2 hasn't print anything we guess your serial port is error.
If the camera is defective , the host_v2 will print"SPI interface Error! " or "Can't find OV2640 module".Please ensure your serial port is right and ensure it is not occupied by other equipment firstly.

jasonc2010, 2018-02-04:
Thank you for the quick response. 

As seen in the image below I have only option for my COM port as I have nothing else connected to my computer.

[image]

IF I keep the serial monitor open in the Arduino programmer I get this:

[image]

Once I close the serial monitor the status goes back to "COM is open!" but just stays in that state.

[image]

jasonc2010, 2018-02-04:
I don't know why, but if I change:

"#else
  Wire.begin();
  Serial.begin(**921600**);"

to:

"#else
  Wire.begin();
  Serial.begin(**250000**);"

And change the baud rate in ArduCAM_Host_V2 from 921600 to 256000 I am now getting:

[image]

And I can take images now. I chose 250000 baud in code because that is the highest baud rate in the Arduino IDE serial port and 256000 is the closest in ArduCAM_Host_V2.

Changing the baud rate in the code to 256000 also seems to work consistently.

supprot, 2018-02-05:
@jasonc2010
Hi,
We konw your problem.
Different boards have different clocks, which will influence your baud rate.
You are using Pro Mini board,so you should choose Pro mini board instead of UNO on Arduino IDE.
Please try it again.

closed 2018-02-06 by jasonc2010