ArduCAM/Arduino #297: ArduCAM_Host_V2 not working?

JSField, 2018-02-05:
I have a ArduCam Mini Ov2640 with ESP8266 NanoV2. I have been able to use the capture and Vid2SD examples and they all work. But I am trying the Host_V2 in Windows 10, and I can connect to the IP, but when I click Capture, no picture or stream is captured. I have SaveImage checked and a path selected, but no file is saved. I can type in the IP address with /stream or /capture in my browser and I get an image/stream from the camera. Should a window open from the Host program and display the feed?

[image]

supprot, 2018-02-05:
Hi,
Very for pur host's bug, Please download  it https://github.com/UCTRONICS/pic/raw/master/ArduCAM_Mini_QT.exe
and copy to your ArduCAM_Host_V2.0_Windows path.

JSField, 2018-02-06:
Hi,
Thank you, this version gives me a picture and a live stream from the camera. But this version does not allow the functions to change the Light or Color or Contrast. Also it doesn't allow to 'SaveImage'. Is there a bug in the Host_V2 program?

supprot, 2018-02-06:
@JSField
Hi，
Yes, at present, our esp8266 demo  don't support setting more functions.
we will update it in the future.