ArduCAM/Arduino #298: OV5642 Mini Plus Capture Time

williambore, 2018-02-07:
Hi, first of all, I am a complete novice when it comes to Arduinos. As part of my university project I need to be able to capture HQ images (1080p/5MP) at about 1fps and save to an sd card on a quadcopter remotely. I plan on activating the camera using the spare sbus outputs on the transmitter. I bought this camera as it was advertised as being capable of doing 5 MP @ 15 fps. I have connected the camera to an Arduino UNO and then to my PC and uploaded the functions example to the board. When using the host app and choosing the 5 MP resolution captures take up to 10 seconds long to process. I am waiting on an SD card breakout board to try the SD card example and was hoping that this results in quicker image captures? Will this solve my problem or do I need additional hardware? 

Thanks in advance,
Will

supprot, 2018-02-08:
@williambore
Hi,
The frame rate of the speed has a lot of to do with the speed of the controller.
If you want to get higher speed , we advice you use STM32 controller.

williambore, 2018-02-08:
Ok so what sort of frame rate can I expect from an Arduino Uno at 1080p / 5MP?

Thanks

supprot, 2018-02-09:
@williambore 
Hi, we don't advice you use arduino uno if you want to get a big resolution .
the speed of frame is very low and can't up to 1 frame.

williambore, 2018-02-13:
Ok, so what do I need to achieve at least 1 fps at 1080p / 5MP?