ArduCAM/Arduino #299: OV5642 No valid JPEG-data

K0M15CH, 2018-02-09:
Hi,

for debugging reasons I changed the example-Code ESP8266_OV5642_Capture to output the Imagedata in Hex using the serial port. Then I tried to save it with a hexeditor as .jpg. I cant open it with any image editing software, although GIMP throws

> Corrupt JPEG data: 253 extraneous bytes before marker 0xc0

This is for every picture the same, but the data differs starting byte 540.
There is a deviation to the JPEG magic number.
It should be
>FF D8 FF E0 ?? ?? 4A 46 49 46 00 01
and it is missing a char and two extra zero's at the end
>FF D8 FF E0 01 04 A4 64 94 60 11

Am I doing something wrong?

A example Image
https://drive.google.com/open?id=1R41wNjCXX-Jqr8UpiPbghQ9-4c97vgqY

My Code
https://pastebin.com/w3SBr2f4

supprot, 2018-02-10:
@K0M15CH 
Hi,
We have check your damage image. the image doesn't hage the end of the jpeg which 
should be '0xFF 0xD9'.
Please ensure your hardware connection is stable .
What's more , From your code , we find you have a delay(10000), we think that is too long time and 
will make the 8266 reboot. you'd better add some yield().

K0M15CH, 2018-02-10:
Ok, I have found my errors. 
1.  
` while ( (temp != 0xD9) | (temp_last != 0xFF)){}`
is insufficent as file-end recognition.

`do{} while();`

much better.

2. Writing bytes as hex in serial is a problem if the value is smaller than 0x10, because serial is shorting the hex to the first half of the byte. An additional zero is needed.

My new code:

>do{
    temp_last = temp;
    temp = SPI.transfer(0x00);
    if(temp < 0x10){
      Serial.print(0);
      Serial.print(temp, HEX);
    }else{
      Serial.print(temp , HEX); 
    }
    delay(0);
  }while (checkForLastBit(temp, temp_last));
  Serial.println();

Complete code, if someone needs it.
https://pastebin.com/ggzsSAmF

closed 2018-02-10 by K0M15CH