ArduCAM/Arduino #301: Camera (OV2640) slows ESP32 WiFi transfer considerably

Lakoja, 2018-02-24:
Hi.
I have an odd problem I have been tracking now for 6 days and found out today that the Arducam plays at least a part in it:

I have an OV2640 Arducam and an ESP32. The latter one transfers the images over WiFi to a browser or controller.

If I connect the Arducam directly to my board and thus the camera is only an inch away from the ESP32 the WiFi transfer of that drops considerably. Especially there is now from time to time a data block that takes seconds to finish transfer.

I tried different ESPs analyzed power supply and fidgeted much with my code. But this problem also appears with simple test code and the camera simply being connected (not initialized).

The problem is however much weaker / nearly gone if I do **not** connect the camera directly to the board but use a 20cm cable instead.
See the two example images below.
This seems to be a problem of the wire lengths though because moving the camera next to the ESP32 while the cable is used doesn't cause any problems.

Does this ring any bells? Because I have no real idea at the moment.

[image]
[image]

supprot, 2018-02-25:
@Lakoja
Hi,
As normal , you'd better connect the Arducam directly to your board,which will get better signal of the image.
But if you connect the arducam with a cable, it maybe influence the signal of SPI transfer but not influence 
the signal of WIFI.

Lakoja, 2018-02-25:
It seems to be indeed an interference problem while the ArduCam is capturing.

A small strategically placed piece of aluminum foil improves the situation dramatically.

[image]

Lakoja, 2018-02-27:
Ok, that was spoken too soon.
I now need to use a bigger shielding and I have serialized camera action with Wifi transfer: Only one of them is active at a certain time.
If I omit either one of them I still have an extremely erratic Wifi connection.

supprot, 2018-02-27:
@Lakoja
It's weird, it might have a lot to do with your floor.Our ArduCAM ESP32 UNO board is no such phenomenon.

Lakoja, 2018-02-27:
I wondered about the common ground, too.
But moving the camera away by 2cm (with a long pin header) and 1cm out of the board area does not help unfortunately.

Lakoja, 2018-06-05:
My observation was probably/mostly faulty:

I now "resolved" the problem by
- moving WiFi transfer code to the main thread
- Using WiFi channel 1 (instead of 8)

Couldn't really see any influence of the camera after that.

(Before that there was however a close correlation...)

closed 2018-06-05 by Lakoja