ArduCAM/Arduino #302: OpenCV won't open Arducam on PC

coldstart01, 2018-02-25:
Dear sir,

Presently I am using the MT9V034 Arducam with USB shield as a camera. I want to open Arducam in the PC from OpenCV. 

Every time I try to open the Arducam in opencv code it fails to initialize at the outset. How do I fix this?

Best Regards

ArduCAM, 2018-02-26:
Did you try this C/C++ code : www.arducam.com/downloads/app/ArduCAM_USB/x86/ArduCAM_Linux_SDK_OpenCV_MT9V034_Demo.zip
Would you please show me your error message?

coldstart01, 2018-02-26:
error message:

Camera could not be opened.

code:
`
#include "opencv2/opencv.hpp"

using namespace cv;

int main(int, char**)
{
    VideoCapture cap(0); // open the default camera
    if(!cap.isOpened())  // check if we succeeded
    { std::cout<<"Camera could not be opened."<<std::endl;
        return -1;}
    else
    {
    	std::cout<<"Camera initialized."<<std::endl;
    }
    Mat edges;
    namedWindow("edges",0);
    for(;;)
    {
        Mat frame;
        cap >> frame; // get a new frame from camera
        cvtColor(frame, edges, CV_BGR2GRAY);
        GaussianBlur(edges, edges, Size(7,7), 1.5, 1.5);
        Canny(edges, edges, 0, 30, 3);
        imshow("edges", edges);
        if(waitKey(30) >= 0) break;
    }
    // the camera will be deinitialized automatically in VideoCapture destructor
    return 0;
}
`

ArduCAM, 2018-02-26:
Please note that it is not UVC camera, so please ues the example we provide: www.arducam.com/downloads/app/ArduCAM_USB/x86/ArduCAM_Linux_SDK_OpenCV_MT9V034_Demo.zip

coldstart01, 2018-02-26:
But one of your examples is from windows VS; can I use this to process global output from the camera, then from OpenCV?

coldstart01, 2018-02-26:
Can you please post link to visual studio binaries/SDK/source?

coldstart01, 2018-02-27:
is this source code adequate

https://github.com/ArduCAM/ArduCAM_USB_Camera_Shield/tree/master/Windows/USBTest/USBTest

or stay w/ link above?

ArduCAM, 2018-02-27:
SDK has proprietary code which is not available. Why not use the example I send you last  time ? www.arducam.com/downloads/app/ArduCAM_USB/x86/ArduCAM_Linux_SDK_OpenCV_MT9V034_Demo.zip

coldstart01, 2018-02-27:
I can with  time -
I'm mainly windows, not so much linux. It will require a large learning curve I believe. Maybe you could give me a hint on how to set it up, and I can take it up from there : )

coldstart01, 2018-02-27:
Also the patch worked on the driver signing. Will I need it with Windows 10? **Is it ok to install Win 10 with this camera?**

ArduCAM, 2018-02-28:
You can use the Win7 driver and install it on Win10, in order to work around the driver signiture, please check: https://www.youtube.com/watch?v=71YAIw7_-kg

coldstart01, 2018-03-28:
Hi,

Does the example that you sent me last time (Linux) require a client-side opencv installation, or are all the necessary opencv files included? I am getting  the following error when executing:

./Test: error while loading shared libraries: libopencv_core.so.2.4: cannot open shared object file: No such file or directory

closed 2018-03-29 by coldstart01