ArduCAM/Arduino #304: OV2640 - will it do 800x600?

jbishop129, 2018-03-13:
Hi,
I have an OV2640 running on the ESP8266 dev kit, and am streaming the video off the http://ipaddress/stream endpoint successfully to my security DVR. I'm using the "ArduCAM_ESP8266_Nano_V2_Capture" sketch as my starting point, and I was able to successfully increase the resolution up to 640x480, but cannot seem to get it up to 800x600 or higher, although the specs on the device indicate it should be capable. I'm fine with reduced frame rates, if I can achieve higher resolution. Currently I am modifying:
``` myCAM.OV2640_set_JPEG_size(OV2640_320x240);```
to:
```myCAM.OV2640_set_JPEG_size(OV2640_640x480);```
in order to achieve 640x480, but I cannot seem to go any higher; the stream just fails to render. 

Would love to know if this can be done!

UCTRONICS, 2018-03-14:
@jbishop129 
Hi,
The higher the pixel value, the larger the amount of data,which maybe cause the 8266 reset. You can try to change 

 while (!myCAM.get_bit(ARDUCHIP_TRIG, CAP_DONE_MASK));

to  

while (!myCAM.get_bit(ARDUCHIP_TRIG, CAP_DONE_MASK)){
yield();
}

jbishop129, 2018-03-14:
Thanks @UCTRONICS , that seems to have done the trick! I was able to push the resolution up to 1024x768, streaming at a blistering 1 fps! ;)
Thanks for the help

closed 2018-03-14 by jbishop129

UCTRONICS, 2018-03-15:
@jbishop129 
Great .
Let me know if you need more help.
Lvbin from UCTRONICS support team.