ArduCAM/Arduino #305: Stream /  capture on external web server

ivywuuu, 2018-03-13:
Hi, I'm using **Arducam Mini  OV2640 2M** and **Adafruit huzzah esp8266**

I can capture or stream on esp8266 server 192.168.4.1 (set esp8266 as Access Point or connect to my home WiFi).  However, I want to capture / stream on my own AWS web server. ( ex:  www.mysite.com)

Any suggestions?

I'm kinda new to IoT stuff so i'm a little bit lost here. 
Thanks for any help in advance ^_^

UCTRONICS, 2018-03-14:
@ivywuuu 
Hi,

You can try configure int wifiType = 0; // 0:Station  1:AP
Which means our ESP8266 work on STA mode and you should also configure the ssid and password in the below code.
//Station mode you should put your ssid and password
const char *ssid = "SSID"; // Put your SSID here
const char *password = "PASSWORD"; // Put your PASSWORD here
Then our ESP8266 will connect your router.But you must ensure your web server and our ESP8266 on the same LAN.

Let us know if you need more help.
Lvbin from Arducam support team.

Mostxclent, 2018-04-10:
Try this for a similar project http://www.instructables.com/id/Arducam-Mini-With-ESP8266-Wi-Fi-Is-Amazing/