ArduCAM/Arduino #308: ArduCAM Mini OV2640 2MP not working on Arduino MKR 1400 GSM (SAMD21 Cortex-M0+ 32bit low power ARM MCU)

Marcos151196, 2018-03-15:
Hi, 

I know that the **Arduino MKR 1400 GSM** (based on **Arduino Zero**) is not among the supported devices, but is there any way to make it work? Could you give me a modified library son I can get it to work?

This is the error I get when I try to compile it:

```
_In file included from /home/marcos/Arduino/ArduCAM_Mini_2MP_OV2640_functions/ArduCAM_Mini_2MP_OV2640_functions.ino:11:0:
/home/marcos/Arduino/libraries/ArduCAM/ArduCAM.h:732:2: error: 'regtype' does not name a type
  regtype *P_CS;
  ^
/home/marcos/Arduino/libraries/ArduCAM/ArduCAM.h:733:2: error: 'regsize' does not name a type
  regsize B_CS;
  ^
exit status 1_
```

Thanks in advance.

supprot, 2018-03-16:
@Marcos151196 
Hi,
You need to know the CPU model of your board and then add the conditional compilation of the model under ArduCAM. H.
For example,
If we Want to support our ArduCAM ESP8266 UNO board  we just add the below codes:

#if defined(ESP8266)
	#define cbi(reg, bitmask) digitalWrite(bitmask, LOW)
	#define sbi(reg, bitmask) digitalWrite(bitmask, HIGH)
	#define pulse_high(reg, bitmask) sbi(reg, bitmask); cbi(reg, bitmask);
	#define pulse_low(reg, bitmask) cbi(reg, bitmask); sbi(reg, bitmask);
	
	#define cport(port, data) port &= data
	#define sport(port, data) port |= data
	
	#define swap(type, i, j) {type t = i; i = j; j = t;}
	
	#define fontbyte(x) cfont.font[x]  
	
	#define regtype volatile uint32_t
	#define regsize uint32_t
#endif	

In the ArduCAM.h file , you will see many platforms we have added.

Let me know if you need more help.
Lvbin form Arducam support team.

Marcos151196, 2018-03-16:
Ok problem solved, thank you so much, you saved me a lot of time.

zubairzulkifli, 2018-07-19:
Hi @Marcos151196  did you successfully connected to the module?  I'm using MKR1400 also. But in my case, it cannot find the OV2640 module because the I2C address is the same like the ATECC508A chips on the board. So both addresses clashes.

I'm trying to do SERCOM porting for the I2C interface by create a 2nd interface but no success until now.

dansebastian1, 2020-05-21:
I am sorry for this stupid question, but I am completely clueless. I'm trying to connect Arducam Mini 5MP OV5642 with MKR FOX 1200, but I don't know where to find the CPU model or the conditional compilation. Could you please explain?