ArduCAM/Arduino #309: OV2640 blocky/corrupt jpg ouput on wifi demo

Ubipo, 2018-03-17:
Hi,
I'm trying to use the ArduCAM_ESP8266_UNO_Capture example with my Adafruit HUZZAH ESP8266 (an ESP12 breakout). I get perfectly fine images for 160x120 and 176x144 but anything up from that is blocky.
I'm using the cables provided with the camera module directly to the breadboard. I also selected the right camera in memorysaver.h.
I've included a corrupted image and a pic of my setup.

Any help is greatly appreciated!

### ArduCAM_ESP8266_UNO_Capture.ino
```c++
// ArduCAM Mini demo (C)2017 Lee
// Web: http://www.ArduCAM.com
// This program is a demo of how to use most of the functions
// of the library with ArduCAM ESP8266 2MP/5MP camera.
// This demo was made for ArduCAM ESP8266 2MP/5MP Camera.
// It can take photo and send to the Web.
// It can take photo continuously as video streaming and send to the Web.
// The demo sketch will do the following tasks:
// 1. Set the camera to JPEG output mode.
// 2. if server.on("/capture", HTTP_GET, serverCapture),it can take photo and send to the Web.
// 3.if server.on("/stream", HTTP_GET, serverStream),it can take photo continuously as video
//streaming and send to the Web.

// This program requires the ArduCAM V4.0.0 (or later) library and ArduCAM ESP8266 2MP/5MP camera
// and use Arduino IDE 1.6.8 compiler or above

#include <ESP8266WiFi.h>
#include <WiFiClient.h>
#include <ESP8266WebServer.h>
#include <ESP8266mDNS.h>
#include <Wire.h>
#include <ArduCAM.h>
#include <SPI.h>
#include "memorysaver.h"
//#if !(defined ESP8266 )
//#error Please select the ArduCAM ESP8266 UNO board in the Tools/Board
//#endif

//This demo can only work on OV2640_MINI_2MP or ARDUCAM_SHIELD_V2 platform.
//#if !(defined (OV2640_MINI_2MP)||defined (OV5640_MINI_5MP_PLUS) || defined (OV5642_MINI_5MP_PLUS) \
//    || defined (OV5642_MINI_5MP) || defined (OV5642_MINI_5MP_BIT_ROTATION_FIXED) \
//    ||(defined (ARDUCAM_SHIELD_V2) && (defined (OV2640_CAM) || defined (OV5640_CAM) || defined (OV5642_CAM))))
//#error Please select the hardware platform and camera module in the ../libraries/ArduCAM/memorysaver.h file
//#endif
// set GPIO16 as the slave select :
const int CS = 16;

//you can change the value of wifiType to select Station or AP mode.
//Default is AP mode.
int wifiType = 1; // 0:Station  1:AP

//AP mode configuration
//Default is arducam_esp8266.If you want,you can change the AP_aaid  to your favorite name
const char *AP_ssid = "arducam_esp8266";
//Default is no password.If you want to set password,put your password here
const char *AP_password = "";

//Station mode you should put your ssid and password
const char *ssid = "SSID"; // Put your SSID here
const char *password = "PASSWORD"; // Put your PASSWORD here

static const size_t bufferSize = 4096;
static uint8_t buffer[bufferSize] = {0xFF};
uint8_t temp = 0, temp_last = 0;
int i = 0;
bool is_header = false;

ESP8266WebServer server(80);
#if defined (OV2640_MINI_2MP) || defined (OV2640_CAM)
ArduCAM myCAM(OV2640, CS);
#elif defined (OV5640_MINI_5MP_PLUS) || defined (OV5640_CAM)
[... 336 lines cut ...]
	//#define OV7660_CAM
	//#define OV7725_CAM
	//#define OV7670_CAM
	//#define OV7675_CAM
  	#define OV2640_CAM
	//#define OV3640_CAM
	//#define OV5642_CAM
	//#define OV5640_CAM 
	
	//#define MT9D111A_CAM
	//#define MT9D111B_CAM
	//#define MT9M112_CAM
	//#define MT9V111_CAM	
	//#define MT9M001_CAM	
	//#define MT9V034_CAM
	//#define MT9M034_CAM
	//#define MT9T112_CAM
	//#define MT9D112_CAM
#endif 

#endif	//_MEMORYSAVER_
```

[image]
[image]

supprot, 2018-03-17:
Dear customer,
Thanks for your business.
Would you like to attached us some pictures ?
The larger the resolution, the larger the image will be. We advice you use the shortest possible connector to try.

Let us know if you need more help.

Best regards,
Lvbin from Arducam support team
 
From: Ubipo
Date: 2018-03-17 18:05
To: ArduCAM/Arduino
CC: Subscribed
Subject: [ArduCAM/Arduino] OV2640 blocky/corrupt jpg ouput on wifi demo (#309)
Hi,
I'm trying to use the ArduCAM_ESP8266_UNO_Capture example with my Adafruit HUZZAH ESP8266 (an ESP12 breakout). I get perfectly fine images for 160x120 and 176x144 but anything up from that is blocky.
I'm using the cables provided with the camera module directly to the breadboard. I also selected the right camera in memorysaver.h.
I've included a corrupted image and a pic of my setup.
Any help is greatly appreciated!
ArduCAM_ESP8266_UNO_Capture.ino
// ArduCAM Mini demo (C)2017 Lee
// Web: http://www.ArduCAM.com
// This program is a demo of how to use most of the functions
// of the library with ArduCAM ESP8266 2MP/5MP camera.
// This demo was made for ArduCAM ESP8266 2MP/5MP Camera.
// It can take photo and send to the Web.
// It can take photo continuously as video streaming and send to the Web.
// The demo sketch will do the following tasks:
// 1. Set the camera to JPEG output mode.
// 2. if server.on("/capture", HTTP_GET, serverCapture),it can take photo and send to the Web.
// 3.if server.on("/stream", HTTP_GET, serverStream),it can take photo continuously as video
//streaming and send to the Web.
// This program requires the ArduCAM V4.0.0 (or later) library and ArduCAM ESP8266 2MP/5MP camera
// and use Arduino IDE 1.6.8 compiler or above
#include <ESP8266WiFi.h>
#include <WiFiClient.h>
#include <ESP8266WebServer.h>
#include <ESP8266mDNS.h>
#include <Wire.h>
#include <ArduCAM.h>
#include <SPI.h>
#include "memorysaver.h"
//#if !(defined ESP8266 )
//#error Please select the ArduCAM ESP8266 UNO board in the Tools/Board
//#endif
//This demo can only work on OV2640_MINI_2MP or ARDUCAM_SHIELD_V2 platform.
//#if !(defined (OV2640_MINI_2MP)||defined (OV5640_MINI_5MP_PLUS) || defined (OV5642_MINI_5MP_PLUS) \
//    || defined (OV5642_MINI_5MP) || defined (OV5642_MINI_5MP_BIT_ROTATION_FIXED) \
//    ||(defined (ARDUCAM_SHIELD_V2) && (defined (OV2640_CAM) || defined (OV5640_CAM) || defined (OV5642_CAM))))
//#error Please select the hardware platform and camera module in the ../libraries/ArduCAM/memorysaver.h file
//#endif
// set GPIO16 as the slave select :
const int CS = 16;
//you can change the value of wifiType to select Station or AP mode.
//Default is AP mode.
int wifiType = 1; // 0:Station  1:AP
//AP mode configuration
//Default is arducam_esp8266.If you want,you can change the AP_aaid  to your favorite name
const char *AP_ssid = "arducam_esp8266";
//Default is no password.If you want to set password,put your password here
const char *AP_password = "";
//Station mode you should put your ssid and password
const char *ssid = "SSID"; // Put your SSID here
const char *password = "PASSWORD"; // Put your PASSWORD here
static const size_t bufferSize = 4096;
static uint8_t buffer[bufferSize] = {0xFF};
uint8_t temp = 0, temp_last = 0;
int i = 0;
bool is_header = false;
[... 312 lines cut ...]
//#define OV5640_MINI_5MP_PLUS
//#define ARDUCAM_SHIELD_REVC	
//#define ARDUCAM_SHIELD_V2
//Step 2: Select one of the camera module, only one at a time
#if (defined(ARDUCAM_SHIELD_REVC) || defined(ARDUCAM_SHIELD_V2))
	//#define OV7660_CAM
	//#define OV7725_CAM
	//#define OV7670_CAM
	//#define OV7675_CAM
  	#define OV2640_CAM
	//#define OV3640_CAM
	//#define OV5642_CAM
	//#define OV5640_CAM 
	
	//#define MT9D111A_CAM
	//#define MT9D111B_CAM
	//#define MT9M112_CAM
	//#define MT9V111_CAM	
	//#define MT9M001_CAM	
	//#define MT9V034_CAM
	//#define MT9M034_CAM
	//#define MT9T112_CAM
	//#define MT9D112_CAM
#endif 
#endif	//_MEMORYSAVER_

Ubipo, 2018-03-17:
Hi,
Here is another link to the photos I shared in my last post:
[https://photos.app.goo.gl/3pynW5QA2e98BGid2](url)
I'm using the wires included with the camera so I'd assume they are short enough.
Can I maybe lower the bitrate of transmission?
I tried changing SPI.setFrequency but that crashed with 'Soft WDT reset'.
Thanks already for the quick response.

supprot, 2018-03-18:
@Ubipo
Please try to reduce 'static const size_t bufferSize = 4096;' You can try change the 4096 to 1024 or2048.

Ubipo, 2018-03-24:
That fixed it! Thanks so much, I'm sorry if this was noted somewhere and I didn't see it.

closed 2018-03-24 by Ubipo

supprot, 2018-03-26:
@Ubipo 
Great to hear you have solved your problem.
Let me know if you need more help.