ArduCAM/Arduino #313: opencv files, version question

coldstart01, 2018-03-29:
Hi,

Does the example that you sent me last time (Linux) require a client-side opencv installation, or are all the necessary opencv files included? I am getting  the following error when executing:

./Test: error while loading shared libraries: libopencv_core.so.2.4: cannot open shared object file: No such file or directory

If are to install OpenCV do we do that with 3.0+ or match the 2.4, as above.

ArduCAM, 2018-03-29:
you need to install the opencv binary library by sudo apt-get install libopencv-dev

coldstart01, 2018-03-29:
I did that and manually checked /usr/lib and /usr/local/lib and no libopencv_core.so.2.4.
I did system search as well for libopencv_core.so.2.4 as well, with negative results.

The system did install 

a/usr/lib/x86_64-linux-gnu/libopencv_core.so.3.1.0
/usr/lib/x86_64-linux-gnu/libopencv_core.so
/usr/lib/x86_64-linux-gnu/libopencv_core.so.3.1

What next?

coldstart01, 2018-03-30:
Should I uninstall the 3.1.0 and replace with the older distro opencv Unix 2.4?

I really could use the help here, I really want to move forward here.

ArduCAM, 2018-03-30:
I think you need to recompile the source code to run the ./Test example. If you link the source code with opencv 3, it won't require 2.4 version.

coldstart01, 2018-03-30:
Awesome, that makes sense.
What compiler/linker did you use? I plan o using LinuxGCC toolchain unless you think it better to use something else.

ArduCAM, 2018-03-30:
There is comments on top of the cpp file about how to compile the code, something like this: g++ MT9V022_demo.cpp -o Test `pkg-config --cflags --libs opencv` -lArduCamLib -lpthread -lusb-1.0 -L. -I. -std=gnu++11
So you need g++ to compile the code.

coldstart01, 2018-03-30:
Things going ok recompiling source code. It appears opencv library files were not installed wioth the orignal sudo command:

you need to install the opencv binary library by sudo apt-get install libopencv-dev

I can link to header files but not link to library files - they don't exist. I' ve done a thorough search--and manually checked /usr

Should I reinstall opencv?

coldstart01, 2018-03-30:
The documentation is quite good, thanks for checking anyway. My only problem rightn ow is my opencv libraries have literally disappeared. If I can find them I can move on.

coldstart01, 2018-03-31:
Hi Lee, me again. A lot of progress. Can you please tell me what to fix ?
directory>./Test

`Device found = 1 
ArduCam_open successful
create capture thread successfully
create display thread successfully
frame available
No protocol specified
Unable to init server: Could not connect: Connection refused

(Display Image:1991): Gtk-WARNING **: cannot open display: :0
`

ArduCAM, 2018-04-03:
I have never seen "No protocol specified" and "Unable to init server: Could not connect: Connection refused" errors, did you modified our example ?

coldstart01, 2018-04-03:
The only change was using opencv 3.x+ rather than the 2.x in example (
you said to recompile source linking 3.x, which I did).
I recompiled using same command line with proper permissions: 

g++ MT9V034_demo.cpp -o Test `pkg-config --cflags --libs opencv` -
lArduCamLib -lpthread -lusb-1.0 -L. -I. -std=gnu++11
That's when  get the error; I haven't changed the .cpp or .h's either.
My goal was just to get them working in linux.
It builds the program, no problems reported.

Additional information: using Ubuntu OS 17.x on 64bit VMWare linux
virtual machine.
Also have launched it in linux-based *Eclipse Photon *64bit; it doesn't
have the same error as above. Error in this IDE
is simply:

Device found = 0
ArduCam_open failed

Links:

  1. https://github.com/ArduCAM/Arduino/issues/313#issuecomment-378228510
  2. https://github.com/