ArduCAM/Arduino #315: ArduCam-Mini OV5642 issue with sd card

chandanbhatia, 2018-04-05:
When I am connecting both ArduCam and sd card on same SPI bus, I am getting spi connection issue, Please suggest.

supprot, 2018-04-05:
@chandanbhatia 
Which hardware do you use? Maybe your SD card  exclusive SPI bus.

chandanbhatia, 2018-04-06:
I am using catalex micro sd card adopter v1.0 with arduino mega. 
I am also feel same, Any solution?

supprot, 2018-04-08:
@chandanbhatia 
You can consider http://www.uctronics.com/arducam-mini-multi-camera-adapter-board-for-arduino-raspberry-pi.html

chandanbhatia, 2018-04-10:
Yes correct. 
I added one circuit using not gate and NPN transistor before MOSI pin input so MOSI of sd card will activated only when CS PIN is low for same module. Using this I can able to use both camera and sd card but faced one issue that complete image was not able to store in sd card. I can able to see image size around 13Kb in serial monitor but in SD card image size was only 2KB.  I am planing to check using tri state buffer ic in place of not gate and NPN.