ArduCAM/Arduino #320: ArduCAM Mini 2MP - class ArduCAM has no member named 'begin'

LThistle, 2018-04-16:
Hi,

I am trying to use the ArduCAM Mini 2MP camera module (OV2640) to take a timelapse. I followed [this](http://www.arducam.com/knowledge-base/mini-tutorial/) guide on the ArduCam website to setup my libraries, and the camera works fine for taking normal pictures when I use the ArduCAM_Host_V2.0_Windows application. 

But what I really want to do is setup the camera to take a picture every 30 minutes, and so I found [this code](https://github.com/sumotoy/ArduCAM/blob/master/examples/mini/ArduCAM_Mini_2MP_TimeElapse2SD/ArduCAM_Mini_2MP_TimeElapse2SD.ino) to take a timelapse. However, when I try to upload the code to my arduino I get the following error:

      TestCap.ino:40: error: 'class ArduCAM' has no member named 'begin'

         myCAM.begin();

Any help would be appreciated.

supprot, 2018-04-17:
@LThistle 
Hi,
Arducam has update the library. Please download the new code from https://github.com/ArduCAM/Arduino/archive/master.zip
You can use this code https://github.com/ArduCAM/Arduino/blob/master/ArduCAM/examples/mini/ArduCAM_Mini_Capture2SD/ArduCAM_Mini_Capture2SD.ino

LThistle, 2018-04-18:
@supprot 
I've changed my libraries and am using the code you referenced. However when I try to connect my SD card module I get an SPI interface error. [this](https://www.amazon.com/Micro-Adapter-Reader-Module-Arduino/dp/B00NNDBIRK) is the sd card module that I'm using, and since it's made for arduino, I assume it should allow SPI Bus Sharing...

It seems the person [here](https://github.com/ArduCAM/Arduino/issues/38) is having the exact same problem that I have. I keep getting the SPI interface error until I unplug the MISO pin of the arduino. If I do this, the error goes away, but then it can't find the SD card module.

supprot, 2018-04-19:
@LThistle 
From your phenomenon, It's probably because of the card slot. As normal, When the SD  CS is high, it will release the spi bus and will not affect other devices access the spi bus.  I suggest you use an oscilloscope to measure the CS signal and  SPI signal.

closed 2018-05-05 by LThistle