ArduCAM/Arduino #321: Windows Host app saved image size

marcogovo, 2018-04-17:
Dear All,

We'are working with the Arducam-mini-2MP (OV2640) and Arducam Library:
if we configure the library to take 160x120 bitmap image, we noticed that in general
each image is about 37kb.
But, if we launch the Windows Arducam host app, with the same configuration (so, 160x120 without any particular special effect) the saved images are about 2kb.

Could someone explian us this?
Is the Host app that, maybe, automatically compress the images received from the arduino?

Thank you very much,
Bye

Marco

supprot, 2018-04-18:
@marcogovo 
Sorry for our ArduCAM_Host_V2's bug. Indeed, The ArduCAM_Host_V2 stored image in jpeg format without BMP.You can use our old software https://github.com/ArduCAM/Arduino/raw/master/ArduCAM/examples/host_app/ArduCAM_host_v1.0.exe

marcogovo, 2018-04-18:
Hi,

Thank you for the response.
There is not a bug in the Host App, my question is only about the image size.
In few words:
-1 ) if we configure only the Arducam library on the Arduino, in order to take 160x120 images in JPEG format, and we save the image for example on a SD card, this image is about 37kb.
-2) Now we launch the Host App on Windows, with the Arduino connected on the serial port, and the Arducam mini example configured with 160x120 image format and jpeg; so the same configuration of the previous test. But in this case, the image saved on the PC by the Host App is about 2kb.

So, our problem is why the image saved by Host App is so small, respect the image produced only if we use the Arduino+Arducam.

I hope my question now is clearer.

Marco

supprot, 2018-04-18:
@marcogovo 
Sorry for my misunderstanding of your meaning. Now I know what you mean.Don't worry and tomorrow
I will help you test and give you a reply as so as quickly.

marcogovo, 2018-04-18:
@supprot thank you very much,

Marco

supprot, 2018-04-19:
@marcogovo 
Hi,
I have tested it and anything is normal. Can you send us your image?

marcogovo, 2018-04-19:
Dear @supprot,

I attached a [.zip](https://github.com/ArduCAM/Arduino/files/1928321/images.zip)  file with the following images:

160x120-arduino.bmp : 38.5 KB
160x120-win-host.jpg: 3.4 KB
160x120-win-host.bmp: 3.4 KB

We verified that the Windows-Host-App-V2 saves images only in JPEG format, infact
the two win-host images (both .bmp and .jpeg) have the same size and are both JPEG Images.

For us this is not a problem, because out intention is to use only an Arduino + Arducam library.
But for now, we are able only to capture BMP images with just Arducam; 
How can we capture and compess the image, obtaining pure JPEG images with Arducam?

marcogovo, 2018-04-20:
Thanks @supprot ,
At this point, how we can configure the Arducam Library to send JPEG images , instead BMP?
Because in the Arducam examples, the Arduino sends over serial always a BMP header + data.

marcogovo, 2018-04-20:
Thanks,

But i don't want to use the Host App,  i only used it for testing the image formats.

What i would to implement now is only with an Arduino + ArduCam, in order to
send over serial JPEG images (not BMP), and receive them with a custom Python script.
I hope to make my situation clearer.

UCTRONICS, 2018-04-20:
@marcogovo 
Hi,
OK, the Arducam and arduino send over serial JPEG images is very easy. You just need to change our video streaming demo.
In our demo , we start capture when the arduino receive 0x10. You can change it for your project.

marcogovo, 2018-04-20:
@supprot ok, i understand.
Give me some time to test the new sketch with our python script, 
and i'll go back to you with a feedback..

Thank you very much,
Marco

marcogovo, 2018-04-20:
@UCTRONICS 
Thank you very much ;)

Marco