ArduCAM/Arduino #327: SD card problem with Arducam 4 cameras adapter board

lesnie, 2018-05-11:
I am trying to get to work 4 Cameras Adapter Board populated with 4 ArduCam mini OV2640, stacked on top of the Arduino UNO board. Everything works with up to 3 cameras. After attaching the last one (4-th) the SD card error keeps happening. The whole thing is powered by the external 5V, 4A stable power source. I tried to switch cameras to check if all modules are working properly. After adding the fourth camera it always fails. Please advice what to do.

UCTRONICS, 2018-05-12:
@lesnie 
Hi,
Please attach me some pictures of your hardware connection in detail

lesnie, 2018-05-12:
Hi,

the problem is that it is a clean thing - Original Arduino Uno board with 4 cams Adapter board on top populated with ArduCam Mini OV264 and properly formatted SD card. Everything works like it should with one, two or three cameras, fails with all four in place. Consistently. I was checking and all camera modules are good, because it doesn’t make a difference which three cameras are connected and at which place. Additional, beefy power (5V, up to 4A) provided through USB on adapter board. After removing one of the arducam modules, not important which one, the SD card can be used without a problem.

UCTRONICS, 2018-05-14:
@lesnie 
We have tested our code and adapter board and it can work fine.
Don't worry and if your hardware is  defective and we will resend you a new one for free.
Please attached me your hardware connection picture.