ArduCAM/Arduino #328: SPI1 interface Error! with ArduCam ESP8266 V2

NC48401, 2018-05-12:
I have confirmed my OV5642 mini plus works on an arduino uno r3 board. I have followed the tutorial on loading the board manager, adding the json url, and set my camera type in memorysaver.h. (Several times.) Upon uploading the file and opening the serial monitor I get the "SPI1 interface Error!" message. Trying various solutions in other posts has not given success. If I comment out the SPI check I can detect the OV5642, the server starts, but I cannot connect with the app. I am using the ArduCam_ESP8266_UNO_Capture example downloaded from github. Here is the screen dump from serial monitor:
ArduCAM Start!
SPI1 interface Error!
Soft WDT reset
ctx: cont 
sp: 3fff1990 end: 3fff1ba0 offset: 01b0
>>>stack>>>
3fff1b40:  3fff0b44 3fff00b8 3fff0b44 40202578  
3fff1b50:  feefeffe feefeffe feefeffe feefeffe  
3fff1b60:  feefeffe feefeffe feefeffe feefeffe  
3fff1b70:  feefeffe feefeffe feefeffe 3fff0b70  
3fff1b80:  3fffdad0 00000000 3fff0b68 40207530  
3fff1b90:  feefeffe feefeffe 3fff0b80 40100718  
<<<stack<<<
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)
load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x0f
csum 0x0f
~ld
Thanks
Sammy

UCTRONICS, 2018-05-12:
Your SPI interface error.
Please check your SPI CS and other signal connection.
Please enable the OV5642_MINI_5MP_PLUS and disable other platform in the memorysaver.h
file

NC48401, 2018-05-12:
UCTRONICS, thank you for the reply. As I stated in my original post, I have checked memorysaver.h,. I am using an ArduCam_mini_5MP_Plus plugged directly into the ArduCam_ESP8266_UNO V2 board. The connections are as good as ArduCam can make them. I have looked at the signal on the CS pin with a logic probe and I do see it toggling. Any other suggestions?

UCTRONICS, 2018-05-12:
@NC48401 
Great to hear you have a logic probe
The SPI detected is that it write 0x55 to 0x00 register and read it again .You can use the logic probe to check the spi signal

NC48401, 2018-05-12:
Since I am getting the CS data and bypassing the SPI check in the sketch allows a successful start would that indicate possibly I have a defective board? I would think the sketch and board should work together if this isn't a common problem you have heard of. They are both ArduCam products.

NC48401, 2018-05-12:
The registry is returning a 0xAA instead of 0x55

UCTRONICS, 2018-05-14:
@NC48401 
It's very strange.
Are you sure you are using 5mp mini plus?
Please attach me your camera picture

NC48401, 2018-05-14:
I would attach a picture but the camera doesn't work, remember? I find that question a little insulting. I can read the words stamped on the boad very plainly. Again, it is an ArduCam_mini_5MP _ Plus. The one thing I failed to supply is that it is REV B.

UCTRONICS, 2018-05-14:
@NC48401 
Sorry and I just want to help you solve your problem as soon as quickly.
From your last email , you  have confirmed your OV5642 mini plus works on an arduino uno r3 board.
So I am sure the camera a is OK. This is indeed a problem with our old version. Please be patient and 
I will test it for you and give you a reply as soon as quickly.

NC48401, 2018-05-14:
Please forgive me for my testy response. I have had a very bad day but that
does not excuse my behavior. Thank you for your quick responses and
assistance.

UCTRONICS, 2018-05-14:
# @NC48401 
I have tested the code and it can work fine.I guess you use our old library. Please try to remove the old
version and download our new library following the video guide https://www.youtube.com/watch?v=n1dDGNpbxGM&feature=youtu.be
[image]

NC48401, 2018-05-14:
Thanks for all your help, but this is the same video that I have followed the procedures through step by step multiple times, removing, deleting, and downloading from github. I think at this point I will call this item unusable and return it. Thanks again for your assistance.

NC48401, 2018-05-17:
After contacting ArduCam customer support it was determined that my Rev B camera was incompatible with the ArduCam ESP8266. Purchase of a new camera solved the problem. Thanks for the help you gave.
Sammy

closed 2018-05-17 by NC48401

UCTRONICS, 2018-05-17:
@NC48401 
You are welcome.
Have a nice day.

mikefile, 2018-06-19:
Hello. I have the exact same problem and I haven't been able to find a solution for a long time. 
I'm using the exact same hardware (except I have anOV5642 ArduCam_Mini_5MP, without the Plus). I cautiously followed all the steps from the tutorial. I installed the most updated library, correctly configured the memory saver (hw platform: OV5642_MINI_5MP, module: OV5642_CAM), I uploaded the ArduCAM_ESP8266_UNO_Capture example, but I always end up with the same error looping in the Serial Monitor:

`ArduCAM Start!
SPI1 interface Error!

Soft WDT reset

ctx: cont 
sp: 3fff1980 end: 3fff1b90 offset: 01b0

>>>stack>>>
3fff1b30:  3fff0b34 3fff00a8 3fff0b34 4020256c  
3fff1b40:  feefeffe feefeffe feefeffe feefeffe  
3fff1b50:  feefeffe feefeffe feefeffe feefeffe  
3fff1b60:  feefeffe feefeffe feefeffe 3fff0b60  
3fff1b70:  3fffdad0 00000000 3fff0b58 402074c8  
3fff1b80:  feefeffe feefeffe 3fff0b70 40100718  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x0f
csum 0x0f
~ld`

I also checked and my camera is Rev C, not B.

I would truly appreciate any suggestion as I need to get this working asap. 

Thanks.

UCTRONICS, 2018-06-20:
@mikefile 
Arducam has update Arducam 5MP camera to 5MP Plus.
If You camera is not the 5MP Plus, Please contact the customer to change your camera to new version.
linda@uctronics.com

NC48401, 2018-06-20:
I am no expert on this by any means but my understanding is that the OV5642_Mini_5MP does not have download buffering of the picture stream. The ArduCam ESP8266 is written around the Plus version which supports the buffering so that could be your problem. Once I upgraded my 5mP_Plus to the rev c  Plus version everything worked as per the tutorial.

mikefile, 2018-06-26:
Hi. Thank you for the help. I ended up solving the problem. There was an issue with the libraries.

@UCTRONICS about the YouTube guide... Is there anywhere where I can get the source code for the Android app that displays's Arducam's video capture? I would really need it for a project that I'm working on that I'm supposed to deliver soon. So I can implement the functionality on my own app. Thanks.

supprot, 2018-06-27:
@mikefile 
Sorry for we don't provide the Android app source code.You can refer to this link to https://github.com/dydwo92/Android-Simple-MjpegViewer

yunfei1994, 2019-04-14:
@mikefile Hi, can you share how you solve the problem? You mentioned "There was an issue with the libraries.", could you elaborate it a bit?

NC48401, 2019-04-14:
Ultimately I found that my problem was my version of the Arducam was to
early for the ESP8266. There had been a firmware update and was not
backward compatible. I wound up buying a new camera and all worked well.
The only other thing I can remember is the CS line was mis-assigned but I
think that was on the UNO. Sorry if this doesn't help.
Sammy

yunfei1994, 2019-04-14:
@NC48401  Hi, thank you for the fast reply, indeed! What I'm using now are B0068 Arducam Mini 5MP module 0V5642 + Adafruit Feather HUZZAH ESP8266. How can I know if my camera is the old version? I purchased it recently, on the package it has a data "20150275".

NC48401, 2019-04-14:
Your camera is the new one as this was a couple of years ago I had bought
mine. What problem are you having? I am no way a programmer but maybe I can
help you.

yunfei1994, 2019-04-14:
@NC48401 Basically same problem as yours, after running the example code "ArduCAM_ESP8266_OV5642_Capture", I got this in the serial monitor:

ArduCAM Start!
SPI1 interface Error!

Soft WDT reset

ctx: cont 
sp: 3ffffd70 end: 3fffffd0 offset: 01b0

>>>stack>>>
3fffff20:  3fff0da8 00000015 3fff10a8 40207ba0  
3fffff30:  3fff0da8 3fff10a8 3fff10a8 40207d85  
3fffff40:  3ffeaad5 00000001 00000010 40207d85  
3fffff50:  3ffeab0e 3fff0da8 3fff10a8 40207db0  
3fffff60:  3fff0da8 3fff10a8 3fff10a8 40207dfc  
3fffff70:  3fffdad0 3fff10a8 3fff10a8 40203160  
3fffff80:  feefeffe feefeffe feefeffe feefeffe  
3fffff90:  feefeffe feefeffe feefeffe feefeffe  
3fffffa0:  feefeffe feefeffe feefeffe 3fff1180  
3fffffb0:  3fffdad0 00000000 3fff1178 40208d14  
3fffffc0:  feefeffe feefeffe 3ffea600 40100739  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
vbb28d4a3
~ld

NC48401, 2019-04-14:
You might check to see that the Arducam CS line is assigned correctly. They
require CS= 16. I do have one running with an ESP8266 but I had to map out
the ESP pinout to the particular board I am using as it was not a one for
one pinout. I think the Arducam references the actual ESP8266 pinout. The
CS line is D0 on my board which is pin 14 of my board. I'll be gone for a
while and will check with you when I get back.
Sammy

yunfei1994, 2019-04-14:
Yes, I connect CS pin of camera to pin 16 on my board. I even changed it to other pins (e.g. pin 2, pin 15 or whatever) in the code and connect to that pin on my board. But still doesn't work.

NC48401, 2019-04-15:
I remember now that what I had to do on my board was use an ohm meter to
find which pin of my board actually connected to pin 16 of the ESP chip as
it wasn't one to one. Are you using the Arducam code for the camera?

UCTRONICS, 2019-04-15:
@yunfei1994 
Please attach me a picture of  your camera.  The new version camera should use the new library.

yunfei1994, 2019-04-15:
@NC48401  @UCTRONICS Thanks for your help. I solved the issue. The reason is that my wiring of the camera and board was wrong (the SPI pinouts).

NC48401, 2019-04-16:
Glad to hear you got it solved. That was what I was suspecting as they tend
to work well. By the way, where are you located? I am in Seattle Washington.
Sammy

yunfei1994, 2019-04-16:
Haha, thanks! I'm in Espoo, Finland.