ArduCAM/Arduino #329: Cannot use OV5642 with Arduino MRK1000

Timmy93, 2018-05-16:
I'm trying to use Arduino MRK1000 with OV5642 ArduCAM-Mini-5MP-Plus.
In order to do so I first loaded the following example: `ArduCAM_Mini_5MP_OV5642_Plus_Functions`

When I compile I get the same error as here: https://github.com/ArduCAM/Arduino/issues/308
So I edited the ArduCAM.h coping the setting of ESP8266:

```
#define MKR1000 1
#if defined(MKR1000)
	#define cbi(reg, bitmask) digitalWrite(bitmask, LOW)
	#define sbi(reg, bitmask) digitalWrite(bitmask, HIGH)
	#define pulse_high(reg, bitmask) sbi(reg, bitmask); cbi(reg, bitmask);
	#define pulse_low(reg, bitmask) cbi(reg, bitmask); sbi(reg, bitmask);
	
	#define cport(port, data) port &= data
	#define sport(port, data) port |= data
	
	#define swap(type, i, j) {type t = i; i = j; j = t;}
	
	#define fontbyte(x) cfont.font[x]  
	
	#define regtype volatile uint32_t
	#define regsize uint32_t
#endif
```
In this way I solved the compilation problems but the SPI does not work.
The sketch give me the following error: `ACK CMD SPI interface Error!` and I have no idea on how to solve this problem.

This is how I connected the cam to the MKR:
```
CAM   ||||   MKR1000

CS             pin 7                
MOSI           pin 8 (MOSI)
MISO           pin 10 (MISO)              
SCK            pin 9 (SCK)            
GND            GND             
VCC            5V
SDA            pin 11 (SDA)
SCL            pin 12 (SCL)           
```

Here is the pinout of the MKR1000 (e.g. pin 7 correspond to D7):
[image]

Timmy93, 2018-05-17:
@UCTRONICS Thank you for your answer.
Currently I'm making a thesis project and I should read the image and send it to a server through a POST request.
Is it possible to read directly the image and send it without using third party software?

UCTRONICS, 2018-05-18:
@Timmy93 

About post server, which is not our mainly business.
You can refer to our Arducam ESP8266 demos from https://github.com/ArduCAM/ArduCAM_ESP8266_UNO/blob/master/libraries/ArduCAM/examples/examples/ESP8266/ArduCAM_ESP8266_UNO_Capture/ArduCAM_ESP8266_UNO_Capture.ino
This demo's mainly work is that the image data will send to web client directly when you visit the 8266 server by your browser.

agilecomputing, 2018-08-16:
@Timmy93 

Hello - I am also trying to get an ArduCAM (in my case, the ArduCAM Mini 2 MP OV2640) to work with an Arduino MKR device. In my case, it is an Arduino MKRWAN 1300 (but I believe the board is substantially the same as the MKR1000).

However, I am curious why you decided to duplicate the definitions from ESP8266 in ArduCAM.H? Why did you choose that instead of duplicating the settings for ARDUINO_SAMD_ZERO? The ESP8266 has a different type of CPU than the MKR1000, right?

Incidentally, I tried what you had suggested (copying the settings from ESP8266) but it causes the board to hang after a reset. Then I have to double-reset it to get back to the bootloader in order to recover it.

Thank you for any insight you can share.

Dickoabc123, 2023-07-09:
Hey guys, did anyone manage to get this working? I'm having the same trouble on my WAN 1310 board.