ArduCAM/Arduino #330: Shield revc+ and genuino1

catosh, 2018-05-24:
Hello, 
I am facing many issues using the OV2640_CAM with arducam revc+ and arduino/genuino unoR3.
JPG images are partially written on the SD card, BMP images show nothing but noise. 
I am just compiling the example ARUCAM_REVC_Camera_playback after editing the memorysaver.h. 
any hint would be gladly appreciated. 
B.R.
Luca. 
[1.BMP.zip](https://github.com/ArduCAM/Arduino/files/2034903/1.BMP.zip)

UCTRONICS, 2018-05-25:
@catosh 

This is caused by the inversion of the high and low bits of pixel point data.
We have update our code please use https://github.com/ArduCAM/Arduino/blob/master/ArduCAM/examples/Shield_V2/ArduCAM_Shield_V2_Camera_Playback/ArduCAM_Shield_V2_Camera_Playback.ino

catosh, 2018-05-25:
Hello UCTRONICS, thanks for your quick reply. 
I downloaded the updated sketch but still not working fine for the BMP.
Some questions here: 
1) the arducam revC+ shield is in the "Shield V2" folder, not in the REVc folder?
2) do I need to update the firmware of the arducam revc shield? If so, how can I do it? 
3) Which debug information could I provide to enable you to help me on this issue? 
4) I had to slow down the spi clock, at 8MHZ i had errors during initialization. Does this provide some useful information for you? 
Thanks and best regards, 
Luca.

UCTRONICS, 2018-05-25:
@catosh 

We have stop product Shield Rev C+,  Please try enable the define ARDUCAM_SHIELD_REVC	
in the memorysaver.h file and try the https://github.com/ArduCAM/Arduino/blob/master/ArduCAM/examples/REVC/ArduCAM_REVC_Camera_Playback/ArduCAM_REVC_Camera_Playback.ino demo.

catosh, 2018-05-25:
Here I am. I still have the noise BMP in the pictures.
I can spot some compilation errors BTW:
`
In file included from Arduino/OV2640_Luca/ArduCAM_OV2640_Digital_Camera2/ArduCAM_OV2640_Digital_Camera2.ino:22:0:
Arduino/libraries/ArduCAM/ArduCAM.h:112:0: warning: "pulse_high" redefined
 #define pulse_high(reg, bitmask) sbi(reg, bitmask); cbi(reg, bitmask);
 ^
In file included from /home/phate/Arduino/libraries/UTFT_SPI/UTFT_SPI.h:189:0,
                 from Arduino/OV2640_Luca/ArduCAM_OV2640_Digital_Camera2/ArduCAM_OV2640_Digital_Camera2.ino:18:
Arduino/libraries/UTFT_SPI/HW_AVR_SPI_defines.h:5:0: note: this is the location of the previous definition
 #define pulse_high(reg, bitmask) sbi(reg, bitmask); UTFT_cbi(reg, bitmask);
 ^
In file included from Arduino/OV2640_Luca/ArduCAM_OV2640_Digital_Camera2/ArduCAM_OV2640_Digital_Camera2.ino:22:0:
Arduino/libraries/ArduCAM/ArduCAM.h:113:0: warning: "pulse_low" redefined
 #define pulse_low(reg, bitmask) cbi(reg, bitmask); sbi(reg, bitmask);
 ^
In file included from /home/phate/Arduino/libraries/UTFT_SPI/UTFT_SPI.h:189:0,
                 from Arduino/OV2640_Luca/ArduCAM_OV2640_Digital_Camera2/ArduCAM_OV2640_Digital_Camera2.ino:18:
Arduino/libraries/UTFT_SPI/HW_AVR_SPI_defines.h:6:0: note: this is the location of the previous definition
 #define pulse_low(reg, bitmask) UTFT_cbi(reg, bitmask); sbi(reg, bitmask);
`

Editing the ArduCAM_OV2640_Digital_camera2, I obtain an incomplete JPG (n.b. I had to change the second bye to 0xD8 in the jpg file -  size is set to OV2640_160x120) :
[image]
Thanks for your support, 
Luca.

UCTRONICS, 2018-05-26:
@catosh 

From the bad image, it maybe due to the speed of the SPI.
You can try to slow down the spi speed by using  SPI.setFrequency(4000000); //4MHz
You can slow down the speed to try.

catosh, 2018-05-26:
Which version of SPI an SD are you using? 
I don't have the SPI.setFrequency function in my libraries availlable:
`
Arduino/OV2640_Luca/ArduCAM_OV2640_Digital_Camera2_edit/ArduCAM_OV2640_Digital_Camera2_edit.ino: In function 'void setup()':
ArduCAM_OV2640_Digital_Camera2_edit:54: error: 'class SPIClass' has no member named 'setFrequency'
   SPI.setFrequency(4000000);
       ^
Multiple libraries were found for "SD.h"
 Used: Arduino/libraries/SD
 Not used: arduino-1.6.12/libraries/SD
`

UCTRONICS, 2018-05-26:
@catosh 
Please use the setClockDivider() to try, You can set  SPI.setClockDivider(8);
Sets the SPI clock divider relative to the system clock. On AVR based boards, the dividers available are 2, 4, 8, 16, 32, 64 or 128. The default setting is SPI_CLOCK_DIV4, which sets the SPI clock to one-quarter the frequency of the system clock (4 Mhz for the boards at 16 MHz).

UCTRONICS, 2018-05-26:
@catosh 
Please try  SPI.setClockDivider(8);
Sets the SPI clock divider relative to the system clock. On AVR based boards, the dividers available are 2, 4, 8, 16, 32, 64 or 128. The default setting is SPI_CLOCK_DIV4, which sets the SPI clock to one-quarter the frequency of the system clock (4 Mhz for the boards at 16 MHz).

catosh, 2018-05-27:
Hello, 
I tried slowing down the SPI speed up to 32- no troubles on the header but still the JPG picture is incomplete. 
I set a new check on the number of pixels transmitted:
`
    bool complete= false; 
    while( (temp != 0xD9) | (temp_last != 0xFF) | (complete!=true)){
    totalbytes++;
    [...]
    if(totalbytes>RGBSIZE){
        complete = true; 
      }
`
This helps mitigating the issue. 
Now my question is: 
how many bytes should be in a JPG frame? 
I set a simple count on a RGB pciture: 
(160L*120L)*4L*3L
160 *120 pixels of 32 bytes each for each channel. 
Is this correct? 
Further hints? 
Thanks and B.R.
Luca.

UCTRONICS, 2018-05-28:
@catosh 
The JPEG image doesn't have ensure number of bytes.
Different image have different bytes.
Each RGB pixels has 16 bits which is rgb 5 6 5