ArduCAM/Arduino #331: Two issues with writing to SD card from OV5642

nurmister, 2018-05-27:
Hello Mr Lee,

I have been attempting to use the OV5642 plus (i.e., with the shield built-in) with the Arduino Uno for a while now. This Arduino is also connected to an SD shield at CS pin 10. The following are pictures of it if you would be interested in taking a look: https://drive.google.com/open?id=1E1QMGzuoA0izEKikjoGX1mqmTEBtyyiA
There is no apparent conflict of MOSI/MISO and other pins between the Arducam module and the data-logging shield.

The description of the problem is this: I tried to run two of the examples from the Arducam library, ArduCam_Mini_5MP_Plus_Multi_Capture2SD and ArduCam_Mini_5MP_Plus_OV5642_RAW. I leave the examples unmodified except for the specification of the CS pin for the SD (I change it from 9 to 10).

When running both of the examples, both the Arducam and the SD card are recognized fine. The issue is that when pictures are written to the SD card, there are one of two problems:
1. While the file is written to the card (as, for example, "1.jpg"), it is 0 bytes.
2. If the file has a size greater than 0 bytes, it cannot be opened by any image viewing software -- it is apparently corrupt.

Do you know what can be done to solve these issues? If you need any more information regarding my setup I can give it to you.

Thank you.

UCTRONICS, 2018-05-27:
@nurmister 
What's the adapter board you are using ?
Can you test SD card slot separately? Does it can work fine?
I advice you use this demo to check the camera separately 
https://github.com/ArduCAM/Arduino/blob/master/ArduCAM/examples/mini/ArduCAM_Mini_5MP_Plus_Video_Streaming/ArduCAM_Mini_5MP_Plus_Video_Streaming.ino
This demo's work is to send the image data to the serial port. You should use our ArduCAM Host V2 to display the image.
https://github.com/ArduCAM/Arduino/tree/master/ArduCAM/examples/host_app/ArduCAM_Host_V2.0_Windows

nurmister, 2018-05-27:
Thanks for the quick response.

I am using the deek-robot data logging shield (some info on its use, and how it differs from the usual adafruit shield: https://forum.arduino.cc/index.php?topic=516145.0).

As for testing the camera separately -- that is a good idea. The only issue is that I am using a Mac, and running the program is not an easy option. Nonetheless, I will try to get access to a windows machine and try things out in that environment. If you have any comment to make on my choice of data logging shield (e.g. if you have had issues with it yourself) feel free to do so.

nurmister, 2018-05-28:
Hello Again,

I tested out the host app, and successfully wrote to my computer through the camera. This means that my data logger is likely the issue. Do you have any recommendations for alternative data logging shields?

Thanks

closed 2018-06-05 by nurmister