ArduCAM/Arduino #332: "Can't find OV2640 module!"

mtronmaddie, 2018-06-01:
Hello,

I'm trying to get an Arducam MINI 2MP OV2640 to work with my Arduino UNO which is connected to a microSD shield. I connected both the microSD-shield and the Arducam to the pins of the Arduino UNO, 

After some trial and error, I am getting the error message "Can't find OV2640 module !" which comes from the following part of the example code:

ArduCAM_Mini_Capture2SD

`void setup(){
      
     Wire.begin();
     
      uint8_t vid, pid;
      uint8_t temp;
      
      Serial.begin(115200);
      Serial.println(F("ArduCAM Start!"));
      //set the CS as an output:
      pinMode(SPI_CS,OUTPUT);
      // initialize SPI:
      SPI.begin();
      while(1){
        //Check if the ArduCAM SPI bus is OK
        myCAM.write_reg(ARDUCHIP_TEST1, 0x55);
        temp = myCAM.read_reg(ARDUCHIP_TEST1);
        
        if (temp != 0x55){
          Serial.println(F("SPI interface Error!"));
          delay(1000);continue;
        }else{
          Serial.println(F("SPI interface OK."));break;
        }
      }
      //Initialize SD Card
      while(!SD.begin(SD_CS)){
        Serial.println(F("SD Card Error!"));delay(1000);
      }
      Serial.println(F("SD Card detected."));

      Serial.print(" vid: ");      
      Serial.print(vid);
      Serial.print(" pid: ");
      Serial.print(pid);
      #if defined (OV2640_MINI_2MP)
        while(1){
          //Check if the camera module type is OV2640
          myCAM.wrSensorReg8_8(0xff, 0x01);
          myCAM.rdSensorReg8_8(OV2640_CHIPID_HIGH, &vid);
          myCAM.rdSensorReg8_8(OV2640_CHIPID_LOW, &pid);
          if ((vid != 0x71 ) && (( pid != 0x1c ) || ( pid != 0x42 ))){
       
            Serial.println(F("Can't find OV2640 module!"));
            delay(1000);continue;
          }
          else{
            Serial.println(F("OV2640 detected."));break;
          } 
        }
      #elif defined (OV3640_MINI_3MP)
        while(1){
          //Check if the camera module type is OV3640
          myCAM.rdSensorReg16_8(OV3640_CHIPID_HIGH, &vid);
          myCAM.rdSensorReg16_8(OV3640_CHIPID_LOW, &pid);
          if ((vid != 0x71) || (pid != 0x4C)){
            Serial.println(F("Can't find OV3640 module!"));
            delay(1000);continue; 
          }else{
            Serial.println(F("OV3640 detected."));break;    
          }
       } 
      #else
        while(1){
          //Check if the camera module type is OV5642
          myCAM.wrSensorReg16_8(0xff, 0x01);
          myCAM.rdSensorReg16_8(OV5642_CHIPID_HIGH, &vid);
          myCAM.rdSensorReg16_8(OV5642_CHIPID_LOW, &pid);
          if((vid != 0x56) || (pid != 0x42)){
            Serial.println(F("Can't find OV5642 module!"));
            delay(1000);continue;
          }
          else{
            Serial.println(F("OV5642 detected."));break;
          } 
        }
      #endif
      myCAM.set_format(JPEG);
      myCAM.InitCAM();
      #if defined (OV2640_MINI_2MP)
        myCAM.OV2640_set_JPEG_size(OV2640_320x240);
      #elif defined (OV3640_MINI_3MP)
        myCAM.OV3640_set_JPEG_size(OV3640_320x240);
      #else
        myCAM.write_reg(ARDUCHIP_TIM, VSYNC_LEVEL_MASK);   //VSYNC is active HIGH
        myCAM.OV5642_set_JPEG_size(OV5642_320x240);
      #endif
      delay(1000);
}`

and from ArduCAM_Mini_2MP_OV2640_functions

`#if defined (OV2640_MINI_2MP)
  while(1){
    //Check if the camera module type is OV2640
    myCAM.wrSensorReg8_8(0xff, 0x01);
    myCAM.rdSensorReg8_8(OV2640_CHIPID_HIGH, &vid);
    myCAM.rdSensorReg8_8(OV2640_CHIPID_LOW, &pid);
    if ((vid != 0xB6 ) && (( pid != 0x00 ) || ( pid != 0x42 ))){
      Serial.print(F("ACK CMD Can't find OV2640 module!"));
      Serial.print("vid: ");
      Serial.print(vid);
      Serial.print("  pid: ");
      Serial.println(pid);
      delay(1000);continue;
    }
    else{
      Serial.println(F("ACK CMD OV2640 detected."));break;
    } 
  }`

I changed the values (vid 0xB6, pid 0x00, and pid 0x42) and problems are still persisting. When I check my serial monitor, all I see are dots. I made a chart of the values I have gotten, and the values I need.

___   |   EXPECT   |   values   |           need   
VID  |  0x26        |   38         | 0x69   105
PID  |  0x41        |   65         | 
PID  |  0x42        |   65         | 0x1C   28
VID  |                 |                |           182
PID  |                 |   0           |             0

@UCTRONICS

UCTRONICS, 2018-06-02:
@mtronmaddie 

The 2MP‘s pid  is 0x42 and vid is 0x26. How do you change it ?
Please try the below code :
  while(1){
    //Check if the camera module type is OV2640
    myCAM.wrSensorReg8_8(0xff, 0x01);
    myCAM.rdSensorReg8_8(OV2640_CHIPID_HIGH, &vid);
    myCAM.rdSensorReg8_8(OV2640_CHIPID_LOW, &pid);
    if ((vid != 0x26 ) && (( pid != 0x41 ) || ( pid != 0x42 ))){
      Serial.println(F("Can't find OV2640 module!"));
      delay(1000);continue;
    }
    else{
      Serial.println(F("OV2640 detected."));break;
    } 
  }

mtronmaddie, 2018-06-04:
@UCTRONICS, I was not changing the values I was getting from the ArduCAM, but I was reading different vid and pid values depending on which sample code I was running.  

The vid values I got were (0x69) and pid (0x1C) from the ArduCAM_Mini_Capture2SD sample code.  When I tried the ArduCAM_Mini_2MP_OV2640_functions code, I got values of (182 decimal) for vid and (0) for pid.

I'm pretty sure I bought the correct camera.  Why would the values be changing?

UCTRONICS, 2018-06-05:
@mtronmaddie 
From the phenomenon, It maybe  because of the bad hardware connection and the driver of I2C .
I advice you test your i2c bus separately. 
Notice: The connection between the UNO and camera shoud be as short as possible.

mtronmaddie, 2018-06-06:
@UCTRONICS, I’m using SPI, not I2C. And my connection is fine!

UCTRONICS, 2018-06-07:
@mtronmaddie 
The code read the camera's ID (pid and vid )through I2C interface.

mtronmaddie, 2018-06-08:
We connected the I2C wires, but the values we're reading for pid & vid are not the expected values for a OV2640 module (Mini 2MP).  Why would the values on our board not match what is expected?