ArduCAM/Arduino #333: ArduCAM_Host_V2.0 problem

giogrimo, 2018-06-05:
Hi,

I bought an ArduCAM Mini 5MP Plus and I connected it to Arduino UNO board. I followed all steps for libraries configuration as stated in https://github.com/ArduCAM/Arduino. When the host app starts, both SPI and OV sensor are correctly detected. 
BUT when I try to change settings (i.e. Resolutions, Color etc.) there is not correspondence between what I selected from the drop-down menu and what the display shows (i.e. if I select BW option, display shows setting to -60 degrees and the pictures is not in BW, etc.).
I tried both Windows and Mac version of the host app finding same problems.

How could I do to solve this problem? 

Thanks in advance 
giogrimo

UCTRONICS, 2018-06-06:
@giogrimo 
Sorry to hear your problem.
Have you upload this demo https://github.com/ArduCAM/Arduino/blob/master/ArduCAM/examples/mini/ArduCAM_Mini_5MP_OV5642_Plus_Functions/ArduCAM_Mini_5MP_OV5642_Plus_Functions.ino

giogrimo, 2018-06-06:
Yes, I uploaded that demo. 

Are you aware about this issue? 

Thanks
giogrimo

UCTRONICS, 2018-06-06:
@giogrimo 
 I will test it

giogrimo, 2018-06-07:
Ok, thanks, I'm waiting your response.

giogrimo

giogrimo, 2018-06-12:
Hi, @UCTRONICS, 
is there any news?

giogrimo

UCTRONICS, 2018-06-12:
@giogrimo 
I think you have not select  5642 in the Pix .
I have tested it and it can work fine
[image]

giogrimo, 2018-06-28:
Hi @UCTRONICS, 
thank you for the support: now it works using Windows platform. 

But iOS interface does not work as before, even if I selected the correct Pix (the resolution is not the one I selected from the drop-down menu). Have you any idea about the reason? 
In addition the iOS version seems very poor about settings (I'm not able to set Function like BW etc.). Does exist a way to manage functions also using iOS version?

Here, some screens about mentioned issues
[image]
[image]

thanks,
giogrimo

UCTRONICS, 2018-06-28:
@giogrimo 
We have not update the Host_v2 mac version. I will ask our engineer to update it .

giogrimo, 2018-06-28:
Ok, thanks.

giogrimo

giogrimo, 2018-07-16:
Hi, do you already update the Host_v2 mac version?

thanks, 
giogrimo

UCTRONICS, 2018-07-17:
@giogrimo 
Hi,
Sorry, our engineer's business is too busy these days. After updating the host_V2，we will announce it on the website.

RolBak, 2022-03-28:
Hi! i am working on a project in with i am using ArduCAM_mini_2MP_Plus for data collection. i had download the libraries and i was able to upload it in Arduino IDE thus i was capable of getting single shoot pictures. But my only problems are the ArduCAM_Host_V2 app menu bottom are overlapping and i don't know how to fix that. and also, when it comes to use the Bluetooth HC-06 the dialogue windows box where the pictures i am capturing is not showing at all. i am desperately seeking for help since my due date of the project is so close. thank you 
[image]