ArduCAM/Arduino #335: what's arducam?

emetss, 2018-06-15:
Hi guys,
simple and stupid question. What's Arducam?
Is it only Arduino shield for camera that can be controlled by Arduino Program? 
Is it Arduino shield with camera that can be mounted on arduino board and controlled by Arduino Program? 
Is it Arduino shield with camera with library (like openCv) that can be controlled via arduino board?

BDCowboy, 2018-06-15:
I have an arudcam mini esp8266 V2 Nano which is essentially the camera module connected to an ESP-12 with an SD card slot. You can capture images and send them vial wifi to your pc or other device. You can also stream video to a wifi connected device.

emetss, 2018-06-15:
thank you very much for your reply. Sorry I'm new here. I have arduino Ramps 1.4. I want to buy a camera that can be triggered through arduino possibly with the highest resolution possible. I searched and found that the ArduCAM 5MP is the highest resolution that Arduino can handle. So what is the set I have to buy? I mean only the one like the one in the link is enough? Or should I buy shield with it?
https://www.watterott.com/de/5-Mega-pixel-Camera-Module-OV5642-1080P-JPEG-Output

Thanks