ArduCAM/Arduino #336: SPI Interface Error

prasuchit, 2018-06-15:
Hi,

I'm having the exact same error "SPI Interface Error" as https://github.com/ArduCAM/Arduino/issues/141

Please find my circuit and code setup:

[image]
[image]
[image]

Kindly help.

prasuchit, 2018-06-16:
Thank you @BDCowboy, I shall try that and get back to you asap.

UCTRONICS, 2018-06-19:
@prasuchit 
Please refer to this schematic diagram 
[image]
You should connect the SPI signal to here .

prasuchit, 2018-06-19:
@UCTRONICS , thanks for the quick response, however, is this the schematic of the arducam board or arduino, I'm confused. The ICSP pin seems to be among those 6 pins at the middle of the arduino mega board, but that wasn't connected in this picture: http://www.arducam.com/wp-content/uploads/2017/12/ArduinoConnection-.jpg

prasuchit, 2018-06-19:
@UCTRONICS , also I found this link about arduino mega connections, but it is very incoherent, it doesn't mention which sketch it is talking about, doesn't talk about the hardware clearly, and is hard to follow in general: http://www.arducam.com/how-to-connect-arducam-shield-to-mega-2560/
Can you please explain the circuit for arduino mega and any changes in the program? I'm using Arducam_mini_lowpowermode example code from the library.

prasuchit, 2018-06-19:
Ok, I figured it out, for anyone who is having this trouble in the future: 
(This was done with Arduino Mega, but I guess the principle is the same for all boards)
Connect Vcc and Ground of the arducam board to the +5V and Ground on arduino.  The CS pin from the arducam board to any digital pin on arduino and look for "const int SPI_CS = 49;" in the code that you are running and change that to that pin number. The rest of the pins, refer to the image:
[image]

coldeli, 2020-10-11:
This worked for me after searching for a really long time. I don't know why arducam wouldn't have a diagram for mega. This worked for me with the 2MP Plus using MEGA

UCTRONICS, 2020-10-13:
Thanks for your great suggestions. We will update our diagram for mega as soon as possible.

ArduCAM, 2020-10-13:
Hello,
Arducam has update document and add diagram for mega2560+ Arducam MINI Camera 
https://www.arducam.com/docs/spi-cameras-for-arduino/hardware-connection-for-mega2560/

Adigoom, 2023-03-07:
how do i change the clock speed when using the arduino uno r3 and arducam mini 5mp plus