ArduCAM/Arduino #339: Arducam OV5642 Mini and Micro SD card module

KlydeLencioco, 2018-06-27:
Hello guys. I am currently working on a project. I am using your camera OV5642 and Micro SD module. The problem is when I test your code Capture2SD, there is always an error "SPI interface error" but when I disconnect the micro sd module, the spi will suddenly be ok. I am thinking that this is because I use multiple devices on the SPI ports. Do you have any fix on this? Both module have different CS pins. I am using Arduino uno. Thanks

UCTRONICS, 2018-06-28:
@KlydeLencioco 
It has a lot to do with the hardware design of your SD card holder. Maybe it monopolizes the SPI bus.

UCTRONICS, 2018-06-28:
You can use our adapter board.
http://www.uctronics.com/media/catalog/product/cache/1/image/9df78eab33525d08d6e5fb8d27136e95/u/4/u4135_4.jpg

KlydeLencioco, 2018-06-28:
is the board with esp8266 with micro sd compatible too? and by the way I need to integrate the device to atmega328p chip, will that also work?

UCTRONICS, 2018-06-28:
@KlydeLencioco 
This is desinged for Arduino UNO board with 328p chip.You can plug it to the UNO board directly.

KlydeLencioco, 2018-06-28:
I know but the requirement for us was to use only the 328p chip not the board. Will that work on the chip alone?

UCTRONICS, 2018-06-28:
@KlydeLencioco 
Of course, but you need connect it through dupont line.

KlydeLencioco, 2018-06-28:
what is the dupont line?

UCTRONICS, 2018-06-28:
https://www.google.com/search?q=dupont+line&tbm=isch&source=iu&ictx=1&fir=B6WaBr5M59DFPM%253A%252C1TWsblrIEDIyoM%252C_&usg=__ICdM2W4Qhec5ddc59wwBb7VtBa0%3D&sa=X&ved=0ahUKEwjVvsitqfXbAhVCqFQKHUgsA60Q9QEINDAB#imgrc=B6WaBr5M59DFPM: