ArduCAM/Arduino #341: ArduCam Mini 5MP Plus - Images

giogrimo, 2018-06-28:
Hi, 

I taken several photos with Mini 5MP Plus and I always see a green and pink shadows (circle shaped) in the image (below some examples). Is that a problem of the cam or is it a setting issue? When I use BW setting, the image is good. 
[image]
[image]

thanks,
giogrimo

gosselid, 2020-03-28:
Hi,
I have the same issue with my arducam. 
Did you find a way to get good color images?

Thanks

UCTRONICS, 2020-03-30:
Hi,
For different lens, due to different optical parameters, It doesn't really match the sensor. It exist lens shading issue. It need local engineer support to solve it .