ArduCAM/Arduino #342: ArduCam Mini 5MP Plus - Low light problem

giogrimo, 2018-06-28:
Hi, 

I tried to take pictures in a low light environment (640x480 resolution and default parameters) and I notice a very high degradation of the image after 10 minutes of full power mode (taking pictures). 
Below, the first photo is taken in a low light environment after 5 minutes of full power mode in a high light environment and seems very good; the second is taken 5 minutes later and the result is a very bright picture. 

Is this effect a result of overheating (the cam at the time was very hot) or is it a software compensation of a dark environment?

[image]
[image]

Again, very thanks for the help,
giogrimo

UCTRONICS, 2018-06-29:
@giogrimo 
There has a lot to do with the automatic exposure. I have done a test is does exist this problems.
You can disable the automatic exposure mode and set the camera to manual exposure.
You can refer to the refer confurgure as below:

                              // Close auto exposure mode
                               uint8_t _x3503;
				wrSensorReg16_8(0x5001,_x3503|0x01);	 
	      			//Manually set the exposure value 
				wrSensorReg16_8(0x3500,0x00);
				wrSensorReg16_8(0x3501,0x79);
				wrSensorReg16_8(0x3502,0xe0);	

[image]

giogrimo, 2018-07-02:
Thanks.
In the ArduCam_Mini_5MP_OV5642_Plus_Functions example, where I can add the code you provided?

UCTRONICS, 2018-07-02:
@giogrimo 
You can add then after the myCAM.InitCAM();

giogrimo, 2018-07-02:
I tried ti follow your indications but I obtained the following error: 

Arduino:1.8.5 (Windows Store 1.8.10.0) (Windows 10), Scheda:"Arduino/Genuino Uno"

C:\Users\Zorro\AppData\Local\Temp\arduino_modified_sketch_125855\ArduCAM_Mini_5MP_OV5642_Plus_Functions.ino:28:1: warning: narrowing conversion of '240' from 'int' to 'const char' inside { } [-Wnarrowing]

 };

 ^

C:\Users\Zorro\AppData\Local\Temp\arduino_modified_sketch_125855\ArduCAM_Mini_5MP_OV5642_Plus_Functions.ino:28:1: warning: narrowing conversion of '196' from 'int' to 'const char' inside { } [-Wnarrowing]

C:\Users\Zorro\AppData\Local\Temp\arduino_modified_sketch_125855\ArduCAM_Mini_5MP_OV5642_Plus_Functions.ino:28:1: warning: narrowing conversion of '196' from 'int' to 'const char' inside { } [-Wnarrowing]

C:\Users\Zorro\AppData\Local\Temp\arduino_modified_sketch_125855\ArduCAM_Mini_5MP_OV5642_Plus_Functions.ino:28:1: warning: narrowing conversion of '248' from 'int' to 'const char' inside { } [-Wnarrowing]

C:\Users\Zorro\AppData\Local\Temp\arduino_modified_sketch_125855\ArduCAM_Mini_5MP_OV5642_Plus_Functions.ino:28:1: warning: narrowing conversion of '224' from 'int' to 'const char' inside { } [-Wnarrowing]

C:\Users\Zorro\AppData\Local\Temp\arduino_modified_sketch_125855\ArduCAM_Mini_5MP_OV5642_Plus_Functions.ino: In function 'void setup()':

ArduCAM_Mini_5MP_OV5642_Plus_Functions:81: error: 'wrSensorReg16_8' was not declared in this scope

       wrSensorReg16_8(0x5001,_x3503|0x01);   

                                         ^

exit status 1
'wrSensorReg16_8' was not declared in this scope

May I failed to insert the code? 

Thanks 
Giogrimo

ArduCAM, 2018-07-02:
Look like you enter the wrong value:  wrSensorReg16_8(0x5001,_x3503|0x01);  which should be  wrSensorReg16_8(0x5001,0x3503|0x01);

UCTRONICS, 2018-07-03:
@giogrimo 
You should rename all the wrSensorReg16_8  to myCAM.wrSensorReg16_8;

giogrimo, 2018-07-16:
Thank you, 

giogrimo